OnionUI/Onion #7: SquirrelJME can not boot any game

maytinhdibo, 2022-07-07:
### Checklist

- [X] I am reporting a bug in the Onion OS
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.1

### Miyoo Firmware Version

220204190828

### Provide a clear and concise description of the issue

SquirrelJME can not boot any game. I tested some game work on [MiY](https://github.com/TriForceX/MiyooCFW) but not work on OnionOS.

### Steps to Reproduce

Put a JAR file (game file for mobile phone) into JAVA folder and test it.

### Provide any additional context or information that may help us investigate your issue

Check Java can work on it.

Olywa, 2022-07-08:
Hi there, you did not review the WIki but ticked the box to say you did ;)

This is already a known issue and is [documented in the Wiki](https://github.com/Sichroteph/Onion/wiki/2.-Onion-Emulators-&-Ports#ports--miscellaneous-emulators):
[image]

You can follow the SquirrelJ2ME developers [progress on Reddit](https://www.reddit.com/r/SquirrelJME/comments/u902o9/squirreljme_progress_has_been_slow_for_the_past_8/).

There is nothing that can be done by the Onion team I'm afraid but once the core is capable of running games then it will be updated in Onion.

Olywa, 2022-07-08:
Digging into this a little more the developer of [SquirrelJ2ME commented on a thread here](https://www.reddit.com/r/RetroArch/comments/nz5kax/comment/ibxei0g/?utm_source=share&utm_medium=web2x&context=3) 28days ago that the core is still not capable of running software. Nothing can be done by the Onion team so closing this issue.

closed 2022-07-08 by Olywa