OnionUI/Onion #8: Myst ME has no cursor in ScummVM 2.6.0 git

dem1980, 2022-07-07:
### Checklist

- [X] I am reporting a bug in the Onion OS
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.2

### Miyoo Firmware Version

220204190828

### Provide a clear and concise description of the issue

When i'm playing with ScummVM (display as v2.6.0 git 22/05/2022 update from OnionOS 3.10.6) i have no cursor in the game Myst Masterpiece Edition. So i can't play the game because i don't seen where is the cursor.

If i copy the file scummvm_libretro.so (v 2.1.1 04/06/2022 from OnionOS 3.9 or 3.10.5) into \RetroArch\.retroarch\cores\ the cursor is back.

### Steps to Reproduce

Load the game Myst Masterpiece Edition from ScummVM from v3.12.2, the cursor will not appear.

### Provide any additional context or information that may help us investigate your issue

scummvm_libretro.so v2.1.1 works
scummvm_libretro.so v2.5.1 (display in menu as 2.6.0 git) is broken for Myst ME

Olywa, 2022-07-08:
Hi there, thanks for reporting and for troubleshooting first. The fact the issue is present in one core version and not another indicates this is an issue with the core itself and not something the Onion devs can 'fix' (that's for the developer of the core) beyond swapping the default Onion core.

The problem with this is that while this particular game benefits from the older core, there maybe fixes and improvements in the newer version that other games/users are benefiting from and could be broken if we were to revert the default.   

We will leave this one open for now in case someone more familiar with Scummvm has anything to share. At least for now you have a workaround for this game, though I appreciate it is not ideal.

dem1980, 2022-07-09:
**Thank you.
I understand.
Stupid Horoscope had built the new version of the core (aka 2.5.1 or 2.6.0 git).
He could built it with differents parameters from the original version 2.1.1, i don't know who build the initial version 2.1.1 for Miyoo Mini (or arm7).

It seems that we can't add issues in his github so i post here :
https://github.com/StupidHoroscope/ScummVM-MiyooMini**

Olywa, 2022-07-09:
Yeah, I think he has been busy with personal/work commitments lately but we can ask him to take a look as soon as he is able.

StupidHoroscope, 2022-07-09:
Hi, yep! Sorry, it is definitely true that I've got a lot of work commitments on at the moment, but was completely unaware that issues had been disabled on my Github, which should now be enabled, sorry about that! If you add an issue there, I'll see if I can get around to it when I have time (Which may not be for a couple of months)

dem1980, 2022-07-09:
Thank you for your answers.
So i published the issue in the StupidHoroscope's github.

Olywa, 2022-07-09:
Thanks @StupidHoroscope I'm gonna close this issue here and leave it with you for when you have some time. 
@dem1980 this seems like a sensible workaround, thanks for sharing (and for taking the time to report on SHs Github 👍

Olywa, 2022-07-09:
Issue now logged on the core developers github: https://github.com/StupidHoroscope/ScummVM-MiyooMini/issues/1

closed 2022-07-09 by Olywa