OnionUI/Onion #10: [BUG] Missing Audio and uncapped FPS for PSX games (also broken shortcuts?)

wychwitch, 2022-07-08:
### Checklist

- [X] I am reporting a bug in the Onion OS
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

PSX games have no audio.

FPS is seemingly unlocked (games run way too fast at the very least).

Also the default shortcuts (such as select + menu) don't work. Other systems work just fine.

### Steps to Reproduce

1. Install onion OS 3.12.2
2. Put the PSX bios in the BIOS folder
3. Test any PSX games 

### Provide any additional context or information that may help us investigate your issue

I've tried both with the BIOS files that came with MIYOO, and ones I downloaded elsewhere.

Personally tried with the following games:

1. chocobo's dungeon 2
2. megaman legends
3. misadventures of tron bonne

Also a few others I cannot recall. They're converted to CHD files, though originally I used M3U playlists that pointed to .cue/bins. Neither works. 

Also, this seems like [the same issue that was posted on reddit 2 months ago.](https://www.reddit.com/r/MiyooMini/comments/ukaux5/no_sound_no_fps_cap_strange_behavior/) I'm not sure if this is this same issue or just the same symptoms however!

Olywa, 2022-07-08:
A clean install of Onion will usually resolve this type of issue but out of interest, had you upgraded from a previous version of Onion and were these games working OK for you on your previous OS?

Could you could try switching profiles (to the 'guest' account) in APP section temporarily and try launching them again just to rule out any local config or save state issues.

wychwitch, 2022-07-09:
> A clean install of Onion will usually resolve this type of issue but out of interest, had you upgraded from a previous version of Onion and were these games working OK for you on your previous OS?
No, this was the first time I had installed Onion, so it started out as a clean install! On the default OS it worked just fine when I checked them out
 
> Could you could try switching profiles (to the 'guest' account) in APP section temporarily and try launching them again just to rule out any local config or save state issues.
WOAH OKAY THAT WORKS??? I'm not sure Why because I hadn't changed any settings before trying out PSX games... I'm guessing it's a local config issue then?? How should I go about resetting it?

wychwitch, 2022-07-09:
Alright, I tried again with a clean install and it works now! I'm not sure why it didn't work the first time around. Anyways I'm closing this, sorry for this!

closed 2022-07-09 by wychwitch