OnionUI/Onion #9: Menu button doesn't work if auto save state is turned off

itsmehemant123, 2022-07-08:
### Checklist

- [X] I am reporting a bug in the Onion OS
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

On tapping the menu button while playing a GBA game using the default core, the game switcher activates. However, if I turn off the auto save state via the retro-arch config, the menu button stops working. If auto save state being on is required for game switcher to function, can the menu button at least exit the game like before?

### Steps to Reproduce

On a fresh install, using the default cores for game boy advance games

- Go to App > RetroArch
- Go to Settings > Saving 
- Turn off `Auto Save State`
- Go back > Go back 
- Go to Configuration File > Save Current Configuration
- Exit RetroArch
- Open any GBA game 
- Tap the menu button, the device vibrates but nothing happens
- Long press menu button, the device vibrates but nothing happens

### Provide any additional context or information that may help us investigate your issue

I did a fresh install of OnionOS 3.12.2. Copied over a few game boy advance roms, and then saw the new functionality with the game switcher. 

However, I wanted to manually create a save states, instead of automatic save states So I turned off the auto save state setting via the retroarch app. Now the menu button does nothing. I've had to close games by long pressing the power button now.

tewald, 2022-07-08:
For me even the long press stops working. I have to use menu + select to exit retroarch.

Sichroteph, 2022-07-08:
[keymon_onion.zip](https://github.com/Sichroteph/Onion/files/9072871/keymon_onion.zip)
This was a safety precaution so you don't accidentally loose data, can you use this keymon instead and see if that solve your issue ?

itsmehemant123, 2022-07-11:
Alright, i renamed `keymon_onion` as `keymon`, and then replaced the miyoo > app > keymon executable with the new one you provided.

Now, when i disable the auto save state, and hit the menu button while playing the game boy adv. game, nothing happens. Its still the same behavior - the devices vibrates, but nothing happens. 

Should i place the executable in that zip in a specific place?

itsmehemant123, 2022-07-12:
Oh wow, this works! 

So now, if I turn off the `Auto save state`, the menu button still triggers the game switcher. So I'm not able to switch games or, exit to menu, which is what i wanted! Thanks!

However, on opening a game via the switcher, it loads the latest _auto saved state_. Is it possible for the game switcher to load the _last saved state_, instead of the last _auto saved state_?

Sichroteph, 2022-07-14:
no, RA doesn't have this feature. Glad it sorts out your issue. It is a custom binary I made for you and I don't plan to add it in further versions as the game switcher has to work with auto save states.

closed 2022-07-14 by Sichroteph

ganburighan, 2022-08-19:
Hi I have the same problem, and the link for the zip.file does not work anymore ...