OnionUI/Onion #12: Game Switch freezes on specific game

dmolon, 2022-07-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

I have a GBC rom named _Tetrix DX (World) (SGB Enhanced) (GB Compatible)_ and the Game Switch freezes when this game is selected. I think it can be because the big file name, maybe.

### Steps to Reproduce

Open the game, close, open another game, close, open Game Switch and try to go back to _Tetrix DX (World) (SGB Enhanced) (GB Compatible)_

### Provide any additional context or information that may help us investigate your issue

Not applicable.

Olywa, 2022-07-11:
@dmolon Thanks a lot for testing (and taking the time to report the issue). 

The developer will take a look as soon as he is able.

dmolon, 2022-07-11:
@Olywa No problem! If you need more tests, just ask me.

Sichroteph, 2022-07-11:
@dmolon Thanks to you I was able to track and fix the bug in v3.12.3. Thanks !

closed 2022-07-11 by Sichroteph