OnionUI/Onion #14: Firmware update

7zxkv, 2022-07-16:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and something is incorrect or missing

### Wiki Section

https://github.com/OnionUI/Onion/wiki/1.-Onion-installation

### Description

hi,
it would be good to add missing information (personally I ask myself the question): can we make a backup of our files in any temporary directory on the sd-card and update the official firmware, or is it the firmware will format the SD card (in fact I don't want to transfer tens of gigabytes to the pc if I simply don't have to) so is there an automatic formatting or Nope? thank you. I think adding the answer in the wiki would be a good thing.

best regards

### Correction

same as description

Olywa, 2022-07-17:
It's an interesting point. You _can_ update the miyoo fw with your (Onion or Stock) folders&files in place on the SD but there is always a (low) risk and if something goes drastically wrong then you _could_ lose everything (even in a temp directory) which is why the fw guide we link suggests to make a backup on your PC.

We don't specifically recommend one way or the other as the firmware does not come from the Onion team (obviously) and we cannot be responsible for negative effects or data loss Though some people have overcome issues with not being anle to upgrade the fw by _only_ having the .img file on the SD Card. 

Miyoo's own instructions](https://lemiyoo.cn/upgrade/675.html) recommend "to use a card with a small memory for upgrading",  which implies one different to the one you use for your roms & files. This is the way I personally do it (using a separate SD Card) just to be safe.

Olywa, 2022-07-17:
Ultimately the official firmware update process is one of the manufacturer so you should follow their own guidance.

closed 2022-07-17 by Olywa