OnionUI/Onion #17: Issue Title

kangonumberone, 2022-07-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

when launching into a pokemon gen 2 game (pokemon gold, silver or crystal on the GBC) from the game switcher (at any point after youve set the clock ingame) the device gets stuck and the only way to fix it is to plug my microsd into my pc to remove the rom before rebooting again

### Steps to Reproduce

Please provide detaiLaunch any gen 2 pokemon game (crystal, gold, silver on GBC)
Set the ingame clock
walk to the large building in the starting town 
talk to the professor
pick any pokemon
leave the bulding
leave the starting town via the left exit and walk out on the route
press the home button to switch games to something else
load that other game
press home button again and switch back to the pokemon gameled steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

-

kangonumberone, 2022-07-16:
i solved it by deleting all save related files for the game, sorry for bothering

Olywa, 2022-07-17:
Thanks for writing back and confirming, glad you were able to resolve.

closed 2022-07-17 by Olywa