OnionUI/Onion #18: RetroArch config file saving fails |

freq-mod, 2022-07-18:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

v.3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

I wanted to turn off auto-incrementation of save state slots. However, RetroArch config saving is impossible, it fails with:

`Failed saving config to /path to retroarch.cfg` It sounds very much like no access to RA directory

### Steps to Reproduce

Umm go to Main Menu -> Configuration File -> Save Current Configuration

### Provide any additional context or information that may help us investigate your issue

Creating a new config file works, but when starting a game, uneditable retroarch.cfg is loaded.

freq-mod, 2022-07-19:
I heard other people reporting this problem on RH discord. I can try wiping SD card and doing clear install. 

> (did you change any RA directories or make other changes?

No

danamkaplan, 2022-07-19:
I had this issue too - it was fixed from using a fresh samsung evo sd card.

Sichroteph, 2022-07-19:
When you play a game, your core settings are merged with the general settings. That is why RA won't allow you to overwrite the general settings in-game. In order to be able to save the general settings, close the content first or launch Retroarch from apps.

Olywa, 2022-07-19:
@Sichroteph Interestingly I _was_ able to overwrite the settings while 'in-game' per my above response, after disabling incremental saves. Maybe it also depends on _which_ setting have been changed?

Sichroteph, 2022-07-20:
@Olywa strange, maybe I missed something then is the inherent config save logic :)

closed 2022-07-20 by Sichroteph

SpyMasterP, 2022-07-22:
If I run the GameBoy from the Consoles and I unset "Keep Aspect Ratio" under Settings > Video > Scaling... and then attempt to save the config I get the Failed to save config. If I run RetroArch under Apps, I find "Keep Aspect Ratio" already unset (and of course I am able to save the config), but what's the point if its not using those settings when launching under GameBoy. If core settings are being merged with the general settings, how is one able to change them? It seems like I will have to find some config file at some unknown location which it uses for the merge and make the changes directly with a text editor. I find this very unintuitive.

SpyMasterP, 2022-07-22:
\Saves\CurrentProfile\config\Gambatte\Gambatte.cfg
\Saves\GuestProfile\config\Gambatte\Gambatte.cfg

             video_dingux_ipu_keep_aspect = "true"

I don't think its a good idea to hard-code overrides like this that the user cannot save changes to via the UI. One users preference is not the same as another.