OnionUI/Onion #22: Issue Title

Gallospacca, 2022-07-19:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V 3.11.2

### Miyoo Firmware Version

20220419 

### Provide a clear and concise description of the issue

I need an option to disable the following feature: '' Battery <= 4%: Your handheld will automatically save your game progression and turn off. ''

Or if you can tell me how to disable it.

My problem is that my Miyoo mini always shows me the battery as low (0%), the rest works perfectly (even the charging times and battery life), but this function prevents me from playing, games always close after a few seconds.

With the standard launcher (without Onion) this does not happen but I always have the hateful low battery icon during gameplay, so I switched to Onion and I must say that apart from this problem it is much better in everything so I would like to be able to use it.

Can you remove that function or make it possible to deactivate and activate it?

It would be of great help to me it has been almost a month that I have been trying to solve this problem.

### Steps to Reproduce

Reproducing this situation is simple, just have the battery at 4% or less and try to start a game.

### Provide any additional context or information that may help us investigate your issue

I tried all the available versions of Onion, up to 3.8.3 I always have the low battery icon on the screen as on the stock launcher, subsequent releases remove the icon but all have the problem of games that close by themselves when the battery is low.

I repeat my situation more accurately:
My miyoo mini and its battery work perfectly, the only problem is that the battery is always read as if it were discharged to 0% (but that's not true).

This Onion Os function prevents me from playing so I need a new version with the addition of an option to disable the function, or an indication on which file to edit and how inside the micro sd to disable it.

Olywa, 2022-07-19:
Hi, well it seems your is a pretty unique case where you have a hardware issue. I'm not sure how integrated the warning is or whether if can easily be disabled so we'll need to wait until the Dev has some time to comment. 

I did just want to highlight that you can source a new V2 battery [from here](https://www.aliexpress.com/item/1005004379169865.html?) I have purchased one of these myself as a spare, it was delivered this week and does come with the correct connector for the Mini already installed.

Sichroteph, 2022-07-19:
It is not planned to change this behaviour but looking at the code, the device won't shut off if you disable the auto saves, maybe you will find a solution in this ?

Sichroteph, 2022-07-19:
Keep in mind that removing this option will also disable the menu button in order for you to not loose progression while playing.

Gallospacca, 2022-07-19:
Thanks to both of you for the answers, I already had in mind to try to buy a battery but since the one I have behaves well (both in duration and in recharge times and also the charging LEDs above the console behave normally), Im not completely convinced that it is defective, I thought that a part of the firmware might have been corrupted (the one inside the console not on the sd card) because when I got the console I used it for a few days with its sd card while I was waiting to receive a Sandisk Extreme pro, one day nothing worked anymore and putting the sd in the pc I saw that in all the folders a lot of very strange files had been generated named with very strange and unreadable characters that occupied several gb, so I formatted the card but from that moment the problem arose, so maybe buying a battery will not solve anything, and I can't upgrade the firmware since the Miyoo already has the latest version and I read online that Trying to downgrade there is a risk of bricking the console.

So since my only problem at the moment is that Onion os function that makes the games close, I thought I'd write here, I'm not an expert but I don't think it's very difficult to disable this thing for those who programmed it, so I'm willing waiting for an answer from the programmers, and who knows maybe in the future this solution could also be useful to someone else.

I tried to disable the autosave and loading of the games but it does not solve anything, and it is very difficult to do because it must be done in every game and as I said they close after a few seconds (sometimes after a second) so I don't have the time to do this, but in a couple of games I succeeded but it doesn't fix anything, I need the console to completely ignore the battery status.

Regarding the saving of games this is something that I wanted to disable anyway, when I open a game I like it to start from the beginning like on a real console and not where I left off, I prefer to rely on normal game saves, so no problem, the big problem is that now I can't play, and after trying Onion, go back to the stock firmware I would be sorry since it's much worse.

As I said I have been trying to solve this thing for almost a month and I have tried everything, in desperation I wrote here, I hope for a solution from the developers (￣ヘ￣;)

Gallospacca, 2022-07-19:
Oh! sorry Sichroteph I only realized now that maybe you are one of the developers!

First of all congratulations for Onion Os, I could only try it a little but from what I have seen it completely transforms the console into something better.

Thanks for your answer, could you please tell me if I can disable this feature by modifying some system file or something like that?
As I said I'm not an expert, but I tried to open all editable files (the ones with the gear) within all the folders but I didn't see anything related to the battery.

Gallospacca, 2022-07-20:
Well the fact that I am trying MiniUI does not mean that the problem with Onion is solved (this thread is about the problem with Onion not what launcher I am trying at the moment), in the future I could try Onion again, of course if I am not helped in any way it is impossible that I can use it.

If, on the other hand, you are offended by the criticisms I made to Onion OS, I apologize but they were sincere criticisms and without having anything against you who created it, indeed if I am allowed to give some advice, I better explain my thoughts: the additions that have been done in Onion Os, such as auto save and auto reload of games, forward key in R1, or close games when battery is low, maybe they can be useful or appreciated by most users, but I think that however they should all be easily disabled by those like me looking for an experience as similar as possible to the original consoles, when I start a game I like to enjoy the eventual inftroduction or title of the game, the fast forward button may be useful to someone but for me it is just a discomfort since I end up pressing it by mistake, and closing a game with low battery can be useful but as you have seen in my case it has totally prevented me from using Onion os, that's why I say that improvements or additions are welcome but they must always be selectable and not forced, that's all.

Greetings and best wishes for your projects.

Olywa, 2022-07-20:
Hi @Gallospacca No need to apologise and critique is always welcome (though best provided on the discord where other users can provide either supporting or differing opinions). All things are considered but development effort has to be balanced with use case as it's such a small dev team. 

Since you battery issue is very much a unique hardware issue, these scenarios cannot always be (easily) catered for. The 4% save/shutdown is a hardcoded safety feature that cannot simply be toggled/made optional without a lot of work as we do not have the Miyoo source code and many features implemented in Onion are done with a lot of time & effort to 'hack' existing binaries. It's a lot more complex than you might imagine. 

I was thinking about your issue and wondered if you had tried running with the battery removed and the Mini only plugged in to USB (if will default to charging screen and you will still need to 'power-on' the mini). I wondered whether, if that works fine this way - it _might_ be an indication that the battery itself is the problem (this is still my gut feeling, the battery has some small charging protection circuit and maybe it is not correctly reporting the voltage to the Mini), it may still be worth sourcing a replacement battery in the hope it resolves that particular issue.

Gallospacca, 2022-07-20:
Thanks for the reply and I apologize again if I have caused someone to feel bad, if I connect the console to the charger everything also works with Onion, so maybe it's the battery or maybe it's the console that reads it wrong, I can't know, but in the future I will try surely to buy a spare battery and in case everything goes well I will keep you updated.