OnionUI/Onion #24: Way to avoid autoloading game by holding a certain button on boot

tarikalmashni, 2022-07-23:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.2

### Miyoo Firmware Version

April 2022

### Provide a clear and concise description of the issue

I've been in a complete softlock inside DosBox Pure (launched Soko-Ban).
It asked to input a key for choosing one of two graphic modes so I pressed menu to bring up the keyboard (took me also to the RA menu of course but I resumed).
However I then was in a complete frozen state as nothing took me out there. No button or combination did anything. Tried Menu, Select+Menu, long hold Menu, Select+Start+L2+R2, Power... nothing. Then I hold Power. That was a big mistake as it created a save state (of that corrupted softlock frozen state) and turned off.
Upon turning back in it instantly booted Soko-Ban again and loaded the state that got me locked again.

My only way of getting out of this without a PC (with which I could have deleted the save states) was to hit Menu, 3x Down and A super super fast after the game auto booted, to close it before the state has been loaded.

I appreciate the auto load feature so I don't wanna turn it off by default. Instead it would be cool that if I hold something like Select or Menu while booting the device, it takes me to Onion's main menu instead of autobooting a game and loading a state.

### Steps to Reproduce

Launch Soko-Ban in DosBox and press Menu (however I wasn't able to reproduce it and the point of this is not to reproduce the bug but to offer an allround solution).

### Provide any additional context or information that may help us investigate your issue

-

Olywa, 2022-07-23:
Hi there, 
OK so from your description, it's not a bug we can easily reproduce but more so what you are asking is a feature request/enhancement. It's not an unreasonable one (I like the idea) but I'm not sure how possible it is.  

On a side note, we really encourage feature requests to be raised through discord (as per the issue template titles and 'New Issue' page). I appreciate the lines are a little blurred with this one as an issue led you to this request but starting the discussion on discord is for your benefit as it gives you an opportunity to garner support for the request from other users and more support means a higher likelihood a feature req will be considered.

We will have to wait until the developers have time to comment.

Olywa, 2022-07-27:
> it would be cool that if I hold something like Select or Menu while booting the device, it takes me to Onion's main menu instead of autobooting a game and loading a state.

Hi @tarikalmashni 
I spoke with the dev about your request but unfortunately it is just not possible at this time. It is something they (including the MiniUI dev) tried to achieve before but if the key is not pressed while the binary is effectively running, it cannot be registered, we can only register keys after boot into Onion. 

Perhaps future improvements with DosBox might help avoid this issue but for now, in Onion, there is little that can be done I'm afraid. If anything changes in the future we may be able to revisit it.

closed 2022-07-27 by Olywa