OnionUI/Onion #25: onion freeze loading screen

mcfly90, 2022-07-24:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

april update

### Provide a clear and concise description of the issue

When exit to main menu of onion from a game and press power button for stand by mode all ok, but when press the power button or menu again for wake the miyoo dont run any game, freeze loading and i have to turn off the miyoo mini

### Steps to Reproduce

1.i play any game
2. exit to main menu onion
3. stand by mode pressing power button
4. wake up miyoo again
5. select another game
6. freeze loading screen
7. turn off miyoo for reset miyoo

### Provide any additional context or information that may help us investigate your issue

Please i hope fix this problem, and maybe and future version provide two versions of onion os, one with overlays function and another without

Olywa, 2022-07-24:
I have tried to reproduce this following exactly your steps several times but I have been unable, everything is working as expected with no freezes. I assumed: 

- for point 2, you are exiting to main menu by pressing 'Menu' button and then 'Start' button?
- for point 3, with standby mode you are doing only a short press on the 'Power' button? 

I tried this with GB. GBA & SNES games.
It sounds like something may be wrong with your installation/setup. Is there a specific system you are having this problem with? Have you changed any Retroarch shortcut keys or settings that could have had an impact? At this stage I can only suggest a reinstalled of Onion.

mcfly90, 2022-07-26:
Can you try wait 5 minutes on main menu after press power button, the screen shut down, press the menu button to wake up and select a new game maybe freeze on loading message

Olywa, 2022-07-26:
> Can you try wait 5 minutes on main menu after press power button, the screen shut down, press the menu button to wake up and select a new game maybe freeze on loading message

I'm not sure why you would press the menu button to wake the Mini (instead of the power button) but I did this anyway exactly as you described, twice in fact (once choosing a game I know I had a save state for and once for a game without). Both times the games loaded as expected without issue. 

Given my findings (and the comment from @[Tortugasss](https://github.com/Tortugasss) ) I can only reiterate that a reinstall of Onion is recommend following the [instructions here](https://github.com/OnionUI/Onion/wiki/1.-Installing-Onion). 

The only thing I can add to this is that if you are using either the SD card or the USB SD card reader that came with the Mini, you should replace them (with a known brand from a reputable supplier) as both are poorly made, very prone to failure and causing corruption.

kirqe, 2022-07-27:
@Olywa, I don't want to create a new issue but I've head this problem 3 times already. The gamestate previews(added in the latest onion) disappear after using the system for a while. 

It just happens randomly so I can't describe the steps thus open a new issue.

Usually when you press the central circle button you get into the game switcher with the screenshots of the latest state
After a while there're no screenshots and it just shows the default image(the one that used during boot with onionos text)

The only thing that helps is reapplying the onionui

sorry for hijacking this issue

Olywa, 2022-07-27:
> I don't want to create a new issue

Please do log it separately (even if you cannot provide steps), your issue is completely different to OP and this one is on the verge of being closed as I cannot reproduce it and it seems like a local problem to the user.

Sichroteph, 2022-07-27:
I was able to reproduce the issue.
Indeed, Eggs implemented his keymon so the menu button is to make a screenshot, no wake up the device.
Thanks for your input, it will be fixed for the new version.

closed 2022-07-27 by Sichroteph