OnionUI/Onion #26: Single press of menu button to Onion Launcher can sometimes cause lock of the Onion Launcher, even on reboot

emperordahc, 2022-07-25:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

During gameplay, I can single tap the menu button, but instead of going back to the Onion Launcher, the exact same game will relaunch, reloading the auto state exactly where the game was, including display of RetroArch loading messages. Further presses of the menu button do the same thing; auto state is saved, but the Onion Launcher never starts, the game just relaunches, reloading the new auto state.

Once this happens, the next time the system is rebooted or the Onion Launcher is manually triggered from Apps or Recent, Onion is perpetually crashed. The screen goes black (backlight still on), the menu button and the power button cause vibration feedback, but nothing ever happens. After many attempts of holding the power button, the system will turn off (sometimes showing the Onion shutdown screen, sometimes just turning off completely). From then on, all reboots show the Miiyoo firmware screen, the Onion boot screen and then directly to a black screen with the same vibration feedback from menu or power buttons.

To fix this issue, on the SD card, the file located at “\Saves\CurrentProfile\lists\content_history.lpl” must be edited by computer; the first item in the “items” list must be removed. The item there is the game that was playing when the menu button issue was triggered.

For instance, last night, “Tetris” for the Gameboy started the crash. This first item entry needed removed:

{
      "path": "/mnt/SDCARD/Roms/GB/Tetris (World).gb",
      "label": "",
      "core_path": "/mnt/SDCARD/RetroArch/.retroarch/cores/gambatte_libretro.so",
      "core_name": "Nintendo - Game Boy / Color (Gambatte)",
      "crc32": "",
      "db_name": ""
},

However, this game and auto save state are completely fine! On reboot, the quick switch Onion Launcher will appear, but not autobooting any game. The “game” that appears first is a screenshot of the game being played when Onion Launcher failed, “Tetris”, but the title will now be the first game in the “content_history.lpl”, in this example it was “Back to the Future III” for the Sega Genesis. That “game” needs cleared by pressing X. After doing so, press Start to exit the Onion Launcher and go locate “Tetris” again, either from Recent or from browsing the Console list. It launches perfectly, using the auto save state. Onion Launcher is now completely back to normal.

### Steps to Reproduce

This event happens after a completely random number of games played (10, 20, 100), not mattering what console is being played or how many times I have played the same game before. I cannot find a pattern at all. This issue started at version 3.12.0 and has continued with each 3.12.X update. I have done many complete wipes and reinstalls of Onion. Using a genuine SanDisk 128GB SD card.

### Provide any additional context or information that may help us investigate your issue

Video: This 2-minute video is a complete noted rundown of symptoms before/after fix: [https://youtu.be/nGc4py-xIYc ](https://youtu.be/nGc4py-xIYc )

Side-Suggestion: a workaround for this (and other possible Onion Launcher issues) is a boot button combo press to wipe all of Onion’s preferences, file history, etc. before the Onion Launcher actually boots.)

Appreciate everyone’s work on Onion; loving my Miyoo!

Olywa, 2022-07-25:
Hi there, thanks for taking the time to report and including the video. It's a strange one and not something I've see personally but as you say it's difficult to find a pattern. 

I wonder, is your content_history.lpl particularly large? Maybe you could try clearing out some of the _older_ history in there to see if this makes any difference?

In any case, for this one we will have to wait until the dev has some time to comment.

emperordahc, 2022-07-25:
> I wonder, is your content_history.lpl particularly large? Maybe you could try clearing out some of the _older_ history in there to see if this makes any difference?

The current content_history.lpl file is 23k with 78 games in the "items" list. I've emptied the list down to a single entry to let it repopulate while trying to nail this down, but I've not seen any pattern to the size of the file and when the crashes start.

Sichroteph, 2022-07-27:
Hello, thanks for your input, onion dev here.
I really want to help you, but for this I need to reproduce the issue. To try to reduce the number of culprits, any chance you use hacked roms, or arcade games ?

emperordahc, 2022-07-27:
If it helps, here is my "content_history.lpl" file after the "Tetris" crash, but before I fixed it: [content_history_lpl.txt](https://github.com/OnionUI/Onion/files/9200462/content_history_lpl.txt)

Sichroteph, 2022-07-27:
I don't think it is the culprit. I tried the file on the same game and it worked just fine.
In the .tmp_update folder is sometimes a file called "cmd_to_run.sh" 
Next time your device freeze at start, can you send it to me ?

SuperMartroid, 2022-08-05:
Hi all, as of the past few days I have also been getting this exact issue randomly. However when following emperordahc's guide to remove the most recent entry from the content_history.lpl file, when I then go to launch the ROM again it refuses to go back to Quick launcher every time. I've tried several times and reboots after removing the entry, but now this game does it every time.

I've attached my cmd_to_run.sh file as you've asked, hopefully you'll be able to ascertain something from that?
[cmd_to_run.zip](https://github.com/OnionUI/Onion/files/9269988/cmd_to_run.zip)

Sichroteph, 2022-08-17:
Thanks for your files @SuperMartroid 
We are still trying to recreate this issue.
Can I ask you something again ? Can you reload the problematic game so it is configured to boot on it, check that the game swticher works fine and send me the new content_history file for comparison ?

Sichroteph, 2022-08-17:
If it happens again, can you check :
- In the tmp_update / romScreens folder, if you can look at the picture of you game without issue
- Go to the play activity app, is everything normal for your gaming time ?

SuperMartroid, 2022-08-17:
Unfortunatley I have since continued to 'mess around' with my Miyoo, and now I don't have the same scenario.
I do strongly believe that this is related to a high number of games in the quick launcher and the content_history file.

I had noticed that the content_history file contains more games than are presented in the quick launcher - it seems the quick launcher stops on a set amount of games.
I tested by continually deleting games at the start of content_history file, and noticed that the quick launcher would then display more of older games (as I made more 'room' by deleting newer titles that were displaying in the quick launcher).

So it seems there is a limit of games that the quick launcher can handle - but the content_history file will continue to log every game, and somewhere the quick launcher can't handle the amount in the content_history file?

I can confirm that deleting a large quantity of games from the content_history file (that I had completed or stopped playing) has now resulted in the quick launcher having no problems at all. So I am 99% sure that it struggles with a large amount of titles (I do not know how many though, sorry!)

emperordahc, 2022-08-17:
Before @Sichroteph asked the last request of me (8th post here, asking me to track down the "cmd_to_run.sh" file), I appeared to have stumbled into a fix for myself, so I haven't been able to reply to the request!

After doing the below; I haven't had the issue happen again, and I've been playing really solid for several weeks. (Normally it would happen almost every day.)

What happened:

1. Got stuck in the Menu button not working loop (ie: while in game, press Menu button, and instead of the Onion Launcher starting, the game reloads)
2. [Did NOT turn off power or try to force start the Onion Launcher via the Miyoo App menu, which would hard crash the system]
3. Held Menu button to exit the game and return to the Miyoo menu
4. Went to Recent icon, pressed "A"
5. Pressed Menu button, selected "Clear list" and confirmed
6. Rebooted Miyoo
7. Everything ran completely as normal

Since doing this, I've played the same games that I had been playing during previous lockups and plenty of new ones. I've played single games for hours on end, and I've played dozens of new games for just a few minutes each. No issues, never manually edited any files.

Hope this helps!

Aemiii91, 2022-09-04:
This is appears to be fixed in v4.0.0. Please comment if the bug persists, and we will reopen.

closed 2022-09-04 by Aemiii91