OnionUI/Onion #28: Libretro Daphne may not have the correct implementation for button presses in game drivers lair2.cpp and cliff.cpp

brandish-transfusible, 2022-07-27:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Retrobreeze noticed that when installing Dragon's Lair 2 and Cliff Hanger on Onion OS that neither responded to button presses. See https://www.youtube.com/watch?v=aFtvL267D6w

Looking into the source https://github.com/libretro/daphne it appears that lair2.cpp and cliff.cpp have unimplemented ed 

unsigned lair2::get_libretro_button_map(unsigned id)
const char *lair2::get_libretro_button_name(unsigned id)
See https://github.com/libretro/daphne/blob/master/daphne/daphne-1.0-src/game/lair2.cpp

Methods were as these are implemented in lair.ccp which drive original Dragon's Lair and Space Ace which receive button presses currently on Onion OS

### Steps to Reproduce

Add a Dragon's Lair 2 rom of choice and the Frame File assets to Onion OS per the model outlined in RetroBreeze's gitlab. Run game and be unable to see any button presses register.

https://gitlab.com/retrobreeze/miyoo-mini-daphne-files

### Provide any additional context or information that may help us investigate your issue

1. It may be that lair2.cpp and cliff.cpp are implemented correctly and do button press logic in a different manner.
2. It may be that there are forks of libretro daphne that other projects build that handle this correctly.
3. It may be useful to try and build with https://github.com/DirtBagXon/hypseus-singe which seems better maintained than Daphne if it is possible to build for the MiYoo Mini. Ark OS which runs on rk3266 and similar devices has been able to build this for non-X86 hardware. 

Discussed this issue with tokofaki in the discord early today.

Olywa, 2022-07-27:
Hi there, just wanted to quickly acknowledge this one quickly, it may be a couple of days before I can take a look and provide a proper response. 

Technically, Daphne is not one of Onions 'supported' systems (per the Wiki and the fact that it's inside 'expert') and the Onion devs are not typically compiling or fixing core/emulator issues or building standalones as it would take time away from Onion development. Ideally we need people from the community to get involved and contribute with that. 

I'll have a chat with @Sichroteph soon to see if he already had some thoughts when you discussed with him.

brandish-transfusible, 2022-07-28:
Thank you for replying to this issue!

Is it the case that Onion OS build libretro Daphne from that repository?

Is there a design consensus that would prevent the build and use of alternatives like https://github.com/DirtBagXon/hypseus-singe ?

Sent with [Proton Mail](https://proton.me/) secure email.

Olywa, 2022-07-29:
I'm honestly not sure where the current build came from, will try to find out but give nit;s one of the 'experimental' cores, I'd be surprised if it was built specifically for Onion.

As to the design consensus on emulators, the Onion team is happy to by guided by the community for things like this. If an alternative is available/built that is mutually agreed to be better that the one currently included then we will happily adopt it. The challenge is having someone outside of the immediate dev team who can build/test these things.

Olywa, 2022-08-01:
Hey there, so I did speak with Toto the other day on this, he thinks the current core _may_ have come from **DJM** so you could try tagging him on the discord, maybe he can help. Another user who I believe was involved in compiling or providing many cores early on is **e1000**. Lastly, **eggs** mentioned to another user (in relation to overlays) that he may have a little capacity, he would have to be interested in this but it would be worth a try asking him for sure. 

I did see you had some discussions already with other users on the discord a few days back, it could be worth asking the question to the above folk to see if anyone is willing to help or provide guidance. The SDL thread in the Miyoo Mini channel is also a very good place to ask questions and is monitored by the devs. 

I’m minded to close this issue, it’s not something the immediate Onion team have the capacity to get involved in and it won’t get much visibility as folk only really come here when they have issues (it will still be visible as unfixed in the ‘closed’ section obviously).

brandish-transfusible, 2022-08-03:
Will follow up. What is the best way to advise you of developments here?

Sent with [Proton Mail](https://proton.me/) secure email.

Olywa, 2022-08-08:
The best method would be posting in the discord and tagging any (or all) of [the core team](https://github.com/OnionUI/Onion/wiki/5.-Acknowledgments). We are all frequently monitoring the reddit sub too so even a post there will likely be seen.

closed 2022-08-08 by Olywa