OnionUI/Onion #30: Charging Battery screen only animating 6 frames

varkanoid, 2022-07-28:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

As mentioned here
[https://www.reddit.com/r/MiyooMini/comments/vlfco3/longer_charge_screen_animation_thanks_to/](url)
and here
[https://www.reddit.com/r/MiyooMini/comments/vkv2o4/comment/idurr0o/?utm_source=share&utm_medium=web2x&context=3](url)

Sichroteph created a new chargingstate file to allow 20 frames instead of 6 but it is still playing only 6 frames.

### Steps to Reproduce

I have created chargeState0.png to chargeState19.png files for my animation. Its only playing 0-5 files. I used the native chargestate file with 3.12.3 and also the one on the link in the google drive but both just allow 6 frames to be played. Also tried the one from the 3.12.0 to 3.12.3 patch and thats the same.

### Provide any additional context or information that may help us investigate your issue

Looked at the src file in 3.12.3 and it only shows 6 images not 20.

	SDL_Surface* video = SDL_SetVideoMode(640,480, 32, SDL_HWSURFACE);
	SDL_Surface* screen = SDL_CreateRGBSurface(SDL_HWSURFACE, 640,480, 32, 0,0,0,0);
	
	SDL_Surface* image0 = IMG_Load("chargingState0.png");
	SDL_Surface* image1 = IMG_Load("chargingState1.png");
	SDL_Surface* image2 = IMG_Load("chargingState2.png");
	SDL_Surface* image3 = IMG_Load("chargingState3.png");
	SDL_Surface* image4 = IMG_Load("chargingState4.png");
	SDL_Surface* image5 = IMG_Load("chargingState5.png");
	
	SetBrightness(8);
	int run = 1;

Olywa, 2022-07-29:
It sounds like the update may not have made it to the release (I will check and see if we can get it into the next one), but that doesn't explain why when adding the shared file manually, you still cannot get it to work. 

Can you confirm for me that [this is the DL link and file](https://drive.google.com/file/d/1C44u4chnvB81ZdM6jI2_0ziMV0aMSDzk/view) you are trying with?

Sichroteph, 2022-07-30:
I think there is a confusion. I made a custom keymon in order to make the charging screen displaying the frames for a longer time. I believe another user made one with more frames but it wasn't me.

closed 2022-07-30 by Sichroteph

varkanoid, 2022-07-30:
Thanks, I got the chargestate file that plays 20 frames now from another OS. Works a treat.