OnionUI/Onion #32: Amiga BIOS path potentially misdocumented

rhester72, 2022-08-01:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and something is incorrect or missing

### Wiki Section

https://github.com/OnionUI/Onion/wiki/2.-Emulators-and-Ports

### Description

Under Amiga (including Amiga CD32), the following note is present:

For maximum compatibility add ALL ‘kickstart roms’ to the /BIOS/Kickstarts folder and ensure they are named correctly (files and folders are case sensitive).

This does not appear to be correct, as they were not found when present in Kickstarts but _were_ found when present in the parent (i.e. /BIOS).

It is unclear whether the error is in the documentation or the actual core behavior.

### Correction

If this is intended behavior, the note should be updated to read:

For maximum compatibility add ALL ‘kickstart roms’ to the /BIOS folder and ensure they are named correctly (files are case sensitive).

Olywa, 2022-08-01:
This is strange, I set up and tested Amiga myself before updating the wiki and my experience was the opposite of yours 😀
Placing the kicks in the root /Bios/ folder, my games failed to launch, moving them into the /BIOS/Kickstarts/ folder resolved this.

I don't doubt you but just to be sure we are talking about **puae** (in main consoles). not the **uae4arm** core in the expert section right?

We'll keep this open and I'll do some tests before updating the Wiki (might be week or so as away at the moment).

rhester72, 2022-08-02:
Yes, puae - I _believe_ it was 4x4 Off-Road Racing (WHD) that I tested with, but I'll go back and do some retesting tonight.  Literally any WHD that requires Kickstart 1.3 should show the behavior I'm talking about - though to be honest I rather wish it *were* in a Kickstarts subfolder to keep /BIOS/ from getting too messy.

Olywa, 2022-08-08:
Back from hols at the w/end but still not gotten around to experimenting with this, it's still on my to do list and will try to have a look later this week 👍

Olywa, 2022-08-15:
Hi again, finally got around to testing this tonight. In short, you were correct and the Wiki has been updated, Thanks again for reporting. 

I'd forgotten how finicky Amiga emulation is with some rom formats, just refusing to run even with the right kickstarts! Just stating here for anyone else that stumbles across this issue. I had the most success with .adf and .lha files. Many of my roms were .ipf which also give the dreaded Kickstart 1.3 error (despite the kick rom being present and same game running in .adf format). 
In removing/moving the kick files around for testing, one issue I had was that my kick34005.A500 was named with a capital K, this could have led to the confusion in the wiki.

closed 2022-08-15 by Olywa