OnionUI/Onion #33: Last update 3.11.3 not convenient

DeepLight01, 2022-08-02:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.11.3

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

Hi guys. Thanks a lot for a wonderful os for this fantastic device.
Just wanted to let you know that last update with the app switcher is not very convenient.
What I did not like.
- Vibration when you press menu button.
- When using save and rewind feature multiple times the game sometimes just closes. It is very inconvenient. On 3.11.2 everything worked much more stable. Games did not end this way. Had to downgrade to 3.11.2.

So. 
- Could you please provide a way to switch this app switcher off?
- Provide a setting for on/off vibration on pressing menu button in the game?

Thanks a lot again for your hard work!!!

### Steps to Reproduce

In game press save and rewind multiple times. The game will eventually close.

### Provide any additional context or information that may help us investigate your issue

Thanks!

Olywa, 2022-08-02:
I think you mean 3.**12**.3
Also, we ask very clearly at the new issue button that feature requests and questions should be directed to discord/reddit. Yours is exactly this and should _not_ have been logged as an issue. You would have received an answer much quicker on the discord).
[image]

Now, with that out of the way, there is already a way to do both of what you ask (as per the release notes), Inside the **\tmp_update\** folder you will find the files: .`noGameSwitcher_` and `.noVibration_` 

Rename both of these files to remove the underscore at the end and those features will be disabled.

closed 2022-08-02 by Olywa

DeepLight01, 2022-08-02:
Thanks a lot!