OnionUI/Onion #37: Add developer documentation

synthead, 2022-08-06:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and something is incorrect or missing

### Wiki Section

Developer documentation

### Description

There is no developer documentation for contributing.  It'd be great to have stuff like this in a doc:

- What are the build requirements?
- How do we set up a build environment?
- How do we get our code on the Miyoo?
- How is the code tested?
- What are the contribution guidelines?

### Correction

n/a

Olywa, 2022-08-08:
I see you found your way to the discord already and have some answers now. There are few people who have the knowledge to document the tools and processes and those that do are busy working to improve the Onion experience, as you've seen ;). 

This is a nice to have but I'm not sure it will happen anytime soon so I'm minded to close the issue for now, unless it's something you want to run with? ;)

Olywa, 2022-08-10:
**A note to other developers**
The Onion team welcomes contributors and collaborators, if you are a developer who is interested in getting involved, please reach out to a member of the [core team](https://github.com/OnionUI/Onion/wiki/5.-Acknowledgments) on the discord, as linked at the [new issue button](https://github.com/OnionUI/Onion/issues/new/choose).

closed 2022-09-04 by Olywa