OnionUI/Onion #44: Libretro ScummVM crashes when loading Eye of the Beholder and Eye of the Beholder II

brandish-transfusible, 2022-08-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Install Eye of the Beholder 1 or 2 to an Mini with Onion and launch the game from the ScummVM Menu and ScummVM will load an report missing or corrupt KYRA.DAT file. ScummVM will then drop into the console. 

### Steps to Reproduce

Create and myoogamelist.xml entry for Eye of the Beholder or Eye of the Beholder 2. Create .scummvm files in each directory to use the scummvm short name for each game so they launch. ScummVM reports missing or corrupt KYRA.DAT file. 

This would appear to be the file built on compilation to drive the kyra engine 

The file in master and explanation doc can be found here.
https://github.com/scummvm/scummvm/tree/master/dists/engine-data 

It appears to be necessary for all games in this scummvm engine. 
https://wiki.scummvm.org/index.php/Kyra

### Provide any additional context or information that may help us investigate your issue

If it would be possible to share the setup for building retroarch for Onion it would be possible to build and test changes to work toward fixing this issue. Same goes for libretro Daphne.

Olywa, 2022-08-10:
Hi there, I replied also on discord a moment ago. Unfortunately, as this is an issue with the core itself, it's not something the Onion team can resolve. We rely on the excellent community & core developers to address these things. 

The ScummVM build is by @StupidHoroscope so if I could ask you to please log the issue on his github for this build I'm certain he will take a look when he has some time. You can do that here: https://github.com/StupidHoroscope/ScummVM-MiyooMini/issues

To answer your question regarding the building retroarch for Onion, it's not so well documented yet but there is good info on the discord, I will tag you now in the right thread so you can take a look there.

Thanks

closed 2022-08-10 by Olywa

brandish-transfusible, 2022-08-10:
Thank you.

Sent with [Proton Mail](https://proton.me/) secure email.