OnionUI/Onion #52: Issue Title

ilmenit, 2022-08-13:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

3.12.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Just a minor thing - among Experimental cores in the Installer there is one listed: mu (N-Gage). mu is Palm OS emulator, not Nokia N-Gage, therefore should be listed as: mu (Palm OS).

### Steps to Reproduce

Open the installers, locate the "mu" core.

### Provide any additional context or information that may help us investigate your issue

-

Olywa, 2022-08-15:
Hi there, thanks for reporting, after a little googling I can also find nothing linking this emulator to N-Gage so it does appear to be a mistake. Thanks for reporting. 

@Aemiii91 seems like a quick fix, renaming the folder (in layer 3) from `mu (N-Gage)` to `mu (Palm OS)`, the label in the config seems to be already correct. Feel free to nominate/assign to someone else ;)

Olywa, 2022-08-19:
@schmurtzm I guess this is now covered in the build for next release and can maybe be closed?

schmurtzm, 2022-08-19:
Yes you can close, this emulator has veen removed.

Olywa, 2022-08-19:
> Yes you can close, this emulator has veen removed.

removed or renamed? ;)
Not sure if you maybe confused this one with Squirrel2ME that we also discussed but this one was just for renaming the folder (to replace N-Gage with PalmOS)

schmurtzm, 2022-08-19:
Woups sorry, it has been renamed to "Palm OS - m515 (mu)"

Olywa, 2022-08-19:
Perfect, thanks for confirming. Closed :)

closed 2022-08-19 by Olywa