OnionUI/Onion #61: favourite file fixer

schmurtzm, 2022-08-16:


schmurtzm, 2022-08-16:
Enormous bug in MainUI : if you add the same rom two times in favorite , your favorite file will be broken.

Schmurtz describ the [steps to repair a file here](https://discord.com/channels/529983248114122762/931367023588569180/998035929627377764)

low_k0 has compiled jq which allow to manage favorites easily"

closed 2022-09-04 by Aemiii91

schmurtzm, 2023-02-16:
Seems fixed on the last Miyoo Mini+ firmware. Let's hope that the same fix will be applied for the Miyoo Mini on a future firmware 🤞

filipmnowak, 2023-05-28:
> Seems fixed on the last Miyoo Mini+ firmware. Let's hope that the same fix will be applied for the Miyoo Mini on a future firmware crossed_fingers

not really: https://github.com/OnionUI/Onion/issues/280#issuecomment-1566008442

schmurtzm, 2023-05-31:
On which beta version did you see that ?

filipmnowak, 2023-05-31:
> On which beta version did you see that ?

mentioned in linked issue. it's `Onion V4.2.0.2-beta`.

schmurtzm, 2023-06-01:
> mentioned in linked issue. it's `Onion V4.2.0.2-beta` / https://github.com/OnionUI/Onion/commit/4c7e2db

It should be fixed in the very last beta. But we have new problems with MainUI and some roms which are not showing.

filipmnowak, 2023-06-04:
> > mentioned in linked issue. it's `Onion V4.2.0.2-beta` / [4c7e2db](https://github.com/OnionUI/Onion/commit/4c7e2db)
> 
> It should be fixed in the very last beta. But we have new problems with MainUI and some roms which are not showing.

i feel like issue or other, similar is still happening.
- i searched for a game via search app.
- found one with same name, added to favorite twice. (in default MAME and "expert" Final Burn).
- i didn't noted a change until reboot. all favorites (around 20) beside one game were gone.
... funnily enough, i sorted favorites via names and... it fixed the problem, but game i added twice was not there...

i will take a look later at the code handing this functionality.