OnionUI/Onion #65: avoid to generate update.log

schmurtzm, 2022-08-16:
Avoid to write all the logs on the SD card, this could slow down some tasks and wear out prematurely the sd card.

This can be done by suppressing the log file on the SD card as on MiniUI.

Remark : Killing the `tee` process which write on the SD card could have some side effects.

See MiniUI updater for details : 
`rm -f "$SDCARD_PATH/update.log"` in updater file. 

Could be documented in dev wiki section to re-activate it because it allows easy debugging for devs without UART.

schmurtzm, 2022-08-19:
https://discord.com/channels/529983248114122762/1007796369785897090/1009209978839642254

https://unix.stackexchange.com/a/367919 

> When you delete the file, the tee process still has it open for writing, which means that the disk space is not handed back to the system. That won't happen until all open file descriptors that references the file are closed.
> 
> It's perfectly OK to write to a file that's been deleted, as long as the file descriptor was allocated before the deletion.
> 
> You will have to restart the pipeline for the file to be recreated, and to allow the space taken by the old (now nameless) log file to be reclaimed.
> 
> To avoid having to restart the pipeline, you may choose to truncate the file, i.e. trim its size down to zero without removing it. This would allow tee to continue appending to the file without having to re-open it.

schmurtzm, 2022-08-19:
the log is generated by the "/customer/main" shell script which run the updater like that : ${UPDATE_TMP_DIR}/updater | tee -a $UPDATE_LOG

Aemiii91, 2023-07-04:
Is is fixed by miyoo (thanks to @schmurtzm) in fw version 20230505 or later.

closed 2023-07-04 by Aemiii91