OnionUI/Onion #67: infoPanel : new features

schmurtzm, 2022-08-16:
- [ ] Be able to navigate backward in the manual ([solution ref](https://discord.com/channels/@me/996705786493407332/1007253567809257603), [ref2](https://discord.com/channels/@me/996705786493407332/1007255418227142676))

Make it possible to accept args :

- [ ]  - for multiple files : `./bin/infoPanel res/page1.png res/page2.png res/page3.png res/page4.png`

- [ ] - for a folder : `./bin/infoPanel -d /mnt/SDCARD/Screenshots`

introkun, 2022-08-16:
`Be able to navigate backward in the manual` will be resolved by #53

introkun, 2022-08-16:
@schmurtzm requirements contain non-final statemens (about navigation, about name of the application).
Why do we need pargs and/or args for images? Current solution in #53 is based on json file with list of images to show. If we want to change this behaviour I need to know what is the final one.

schmurtzm, 2022-08-20:
@introkun, I have updated the description, I'm not a specifications writer 😉  but it will be probably more clear like this.

introkun, 2022-08-20:
>  "-j" or "--images-json" --> load images from a json
Lets add format of the json for this argument:
```
{
  "images": [
    { "path": "./page1.png" },
    { "path": "./page2.png" },
    { "path": "./page3.png" },
    { "path": "./page4.png" }
  ]
}
```

introkun, 2022-08-21:
#105 resolves navigation by left/right and exit by the menu button

introkun, 2022-08-21:
Image browsing in directory will be implemented in #106

introkun, 2022-08-27:
"-s" or "--show-theme-controls" --> show themed header/footer/navigation buttons/image number
"-d" or "--directory" --> load all images of a folder (ordered alphabetically)
implemented by #106

closed 2022-09-04 by Aemiii91