OnionUI/Onion #68: Convert activity databases into json

schmurtzm, 2022-08-16:
Due to eventual file corruption we could convert activity databases into json instead of SQL lite.

Schmurtz question : is there so much often corruption ?

schmurtzm, 2023-03-27:
An alternative to avoid DB corruption without switching to SQLite could be done in this way : 

> 0) Record the total played time in a global variable
> 1) Add the time like usual, but them record the db in saves/playActivity_tmp.db
> 2) Sync, reopen the file and read the total time.
> If the new total time is >= to the global variable, we move playActivity_tmp.db to playActivity.db

**But switching to SQLite would also make the update much smaller and faster, so there a better chance the play time update will succeed (instead of just discarding it)**
SQLite would make it more efficient as well.. no more "loading the entire db to add a couple of minutes to one game"

And the new db should contain the entire path to the rom (it can be relative to /mnt/SDCARD/Roms though).

schmurtzm, 2023-04-12:
PR done by nilobject : https://github.com/OnionUI/Onion/pull/751

Aemiii91, 2023-07-04:
This is finally done in:
- #956

closed 2023-07-04 by Aemiii91