OnionUI/Onion #73: pac-man : enhancements

schmurtzm, 2022-08-16:
- [ ] Don't go to next option when switching on or off

- [ ] create a flag (or an arg for pac-man) to disable B / quit on pac-man during install process ([discord conversation](https://discord.com/channels/529983248114122762/1007796369785897090/1008314909366419516))
- [ ] A quit popup : « `X modifications have been made. Are you sure you want to exit without applying ?` »
- [ ] And in small at bottom at the msgbox : « `You can select your packs at anytime by launching app -> pac-man »` "
- [ ] rename folders layer1 ,2,3 to "consoles" , "apps" and "experimental" for more understandable source files in the repo

Aemiii91, 2022-08-18:
We can't simplify the source path. It needs to have this structure to allow for apps to add other folders than their `App` folder. This is for example used by `Search` (`App/SearchFilter` and `Emu/SEARCH`) and `Music player` (`App/Gmu` and `Media`).

schmurtzm, 2022-08-19:
We still have a problem with clock.

On a fresh installation the app clock is already present but when you go in pacman it is displayed not installed
Speaking about that, 🤔  "Set emulation time" in pacman. Can we think to a better name ?

I propose "clock" as in the app list ?
And as legend in Apps list : "Set the current virtual date" ?

Aemiii91, 2022-08-20:
Clock, Commander, 240pSuite, and Gmu are now removed during install.