OnionUI/Onion #86: Manage failed game resume on boot

schmurtzm, 2022-08-17:
Sometimes Onion's users have issues on resume : 
https://www.reddit.com/r/MiyooMini/comments/wqhi21/onionos_stuck_on_game_loading_screen/
https://discord.com/channels/529983248114122762/931367023588569180/1009362145705725973

>  we could implement in the keymon: a timed check to see if RA is still running after terminate_retroarch is called, and then send SIGKILL instead (only when holding, I think would be best)

Sichroteph, 2022-08-17:
They are actually two separate issues.
The first one can be recreated, seems related to a failing of the resume function in MainUI
The second one seems to indicate something wrong with the game switcher when processing the last game played. (It can be its image, its gaming time)

Aemiii91, 2022-09-04:
This should've been fixed for now. Reopen if the issue persists.

closed 2022-09-04 by Aemiii91