OnionUI/Onion #88: Wiki Update download links for Third Party Apps

nettema, 2022-08-18:
### Checklist

- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and something is incorrect or missing

### Wiki Section

6. Third Party Apps

### Description

Download links in Third Party Apps are leading to non existing pages.
For example download [link](https://github.com/Sichroteph/Onion/files/9037662/clock_sdl_app.zip) for "Set Time" app return "Not Found" page.

### Correction

Reupload apps or add correct download links.

nettema, 2022-08-18:
works with vpn

nettema, 2022-08-18:
Only this apps are not dowloadable:
https://github.com/OnionUI/Onion/wiki/6.-Third-Party-Apps#set-time-credit-eggs
https://github.com/OnionUI/Onion/wiki/6.-Third-Party-Apps#boot-logo-changer-credit-eggs
https://github.com/OnionUI/Onion/wiki/6.-Third-Party-Apps#sort-favorites-alphabetically-credit--zebramonkeylizard--olywa123

Ports and music player are dowloading properly.

Olywa, 2022-08-19:
Thanks for reporting, I've just fixed these quickly. That said we probably need to revisit them. 

If you are interesting in boot logo changing, I recommend [starting here](https://www.reddit.com/r/MiyooMini/comments/vty9qr/easy_logotweak_v20_easier_to_navigate_between/) 
Favourite Sorting options will also be included by default in the upcoming release, I think clock adjustment also.

closed 2022-08-19 by Olywa