OnionUI/Onion #91: Ignore dirs starting with a dot in pacman

schmurtzm, 2022-08-19:
Initial problem : ._ mac files impacts Package Manager -> [see this issue on reddit](https://www.reddit.com/r/MiyooMini/comments/uy3iyv/onionos_installer_showing_dot_files_macos/)

Solution : ignore dirs starting with a dot in pacman

2 objectives : 

  - get ride of automatic creation of files from macs
  - be able to hide / desactivate some cores in pac-man (and we add a little sentence in the wiki, something like : )

> deprecated cores are disabled in pac-man by adding a "." prefix

schmurtzm, 2022-08-19:
> The || should be && !
>  we just need to delete len > 2 ||

introkun, 2022-08-19:
The issue is not only in PacMan
[image]

PR #94 (2022-08-20): pacman updates
* fixes #91 (exclude Mac's dot items)
* fix compiler warnings
commit e18d2315 (2022-08-20): pacman updates (#94)

* pacman updates
* fixes #91 (exclude Mac's dot items)
* fix compiler warnings

* * removed excessive filename checks
closed 2022-09-02 by Aemiii91