OnionUI/Onion #93: Apps folders renaming

schmurtzm, 2022-08-19:
2 possibilities : 

---
**1.**
- [ ] - [ ] To distinguish native apps from third party apps theoretically `PlayActivity`, `ThemeSwitcher`, `Guest_Mode `and `SearchFilter` should be prefixed with `Onion_` .
- [ ] Rename `The_Onion_Installer` into `Onion_Installer` too (To match with `Onion_Manual` , `Onion_Tweaks `)

---
**2.**
- [ ] Or easier : we remove `Onion_` prefix everywhere !

Aemiii91, 2022-08-19:
If we're renaming them they should just be the same as their labels.

Aemiii91, 2022-08-19:
The reason for *not* renaming is to have them overwrite old apps on install. And have the old apps still be uninstallable by pacman.

Aemiii91, 2022-08-19:
`The_Onion_Installer` becomes `PackageManager` !

closed 2022-09-02 by Aemiii91