OnionUI/Onion #96: app idea : random game picker

schmurtzm, 2022-08-19:
A shortcut to launch a game randomly ? 
An app for it which could be launched by the new custom shortcut from "Tweaks" app ?

It could be with more advanced features : 
- you launch the random app, it displays a game name and his thumbnail preview (the one from Imgs folder) , press A/Start to start the game  , B to pick another one, menu to exit.
- Toto suggest random little preview videos (when you see a good one you press A to start playging (simplemenu already has this feature ?)

schmurtzm, 2022-09-02:
Requested by users here  : 
https://www.reddit.com/r/MiyooMini/comments/x34fy6/onionos_random_game/

themarkappleby, 2022-11-22:
Its a bit of a hack, but I created a custom Onion OS app to try to tackle this: https://github.com/themarkappleby/onion-os-random-rom. Your mileage may vary. Hopefully we can get official support for this feature in the future though.

scalp42, 2022-12-03:
@themarkappleby thanks for the quick app either way!

It'd definitely be great to have the randomize button appear *per* console (but maybe still have a "global" mode from top `menu.)`

closed 2023-01-17 by Aemiii91