OnionUI/Onion #107: Manage CPU consumption of Onion's Apps

schmurtzm, 2022-08-22:
CPU consumption report  : 
(Checked when it is solved /optimized)

-----------------------------------------

- [ ]  - themeSwitcher : 50% -> too much 🔥 
- [ ]  - infoPanel : 50% -> too much 🔥 
- [ ]  - Search: 50% -> too much 🔥 

-----------------------------------------

- [ ]  - playActivityUI : about 22%, seems a lot for a static UI ?
- [x]  - gameSwitcher : 30% when 0 game in list , seems a lot for a static UI ?
- [ ]  - clock: 26% , seems a lot for a static UI ?

-----------------------------------------

- [ ]  - installUI : about 30% probably due to SDL animation at the bottom ?
- [ ]  - chargingState   :  about 30-35% during animation, 0% when screen off

-----------------------------------------

- [x]  - keymon & batmon : 0% 👍🏻
- [x]  - packageManager  : 0% 👍🏻 
- [x]  - Tweaks   : 0% 👍🏻 

-----------------------------------------

- [ ]  - bootScreen : too fast to really know...

-----------------------------------------

Aemiii91, 2022-08-22:
`installUI` and `chargingState` are OK, they have animations.

`bootScreen` doesn't have a main loop, it's exits immediately after drawing to screen once.

schmurtzm, 2022-08-22:
infopanel is fine when we put a sleep into the main while (just tested).

Aemiii91, 2023-07-04:
Search improved by 486d2f10c133733a4250f090bc5c04ccdf452c93

schmurtzm, 2023-11-10:
Theme Switcher still requires to be fixed in 4.2.3 ;)

JakiShen, 2024-01-14:
Add Gamelist.xml file generator please

MattCatz, 2024-08-23:
Sorry to poke an old issue.

1. Any reason to not to enable compiler optimization by default? i.e. at least `-Og` ( I know `-Os` is enabled for a few of those ). Also enabling LTO for that cross compilation unit optimization?
2. With Clock specifically, the main event loop seems to run ungoverned. Could you save a bit of cycles by adding a little bit of delay in there and/or only redrawing if the time changes or you get a user input? 

I'd be happy to test if you shared your test procedure.