OnionUI/Onion #121: Issue Title

bobotrax, 2022-09-03:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

Onion v4.0.0-rc

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Since the upgrade to 4rc it is no longer possible to change the language in retroarch (settings>user>language).
In previous versions it worked.
This is really annoying for me.

### Steps to Reproduce

In retroarch open the settings menu choose user, there is no more language option.

### Provide any additional context or information that may help us investigate your issue

None.

Olywa, 2022-09-03:
Thanks for taking the time to report, this will be investigated 👍

Aemiii91, 2022-09-03:
This has been fixed in [2c34d39](https://github.com/OnionUI/Onion/commit/2c34d39f65e9179599d1b4695765ff371a64c921). Somehow the `LANGEXTRA` flag had been disabled. Fix will be included in v4.0.0[-rc2].

schmurtzm, 2022-09-04:
Hi, I recommend to keep english only by default.
To have multilang we have to change the main binary  of RetroArch with a bigger one.
If we take the multilang, retroarch consumes 53MB of ram memory instead of 48MB (and we have only 128MB on this device...).

Aemiii propose to implement a toggle in Tweaks -> Advanced

bobotrax, 2022-09-04:
Hello,
 In version V3.12.3 it worked well and Retroarch consumed 51MB, there was 60MB free.
[image]

In version V4.0.0-rc Retroarch consumes 48MB and there is 67MB free.
[image]

Since V4 rc we have gained 6-7MB of free ram.
It's up to you but not everyone is English speaking. After a multilanguage with a Tweaks it's good too.

closed 2022-09-08 by Aemiii91