OnionUI/Onion #122: “Filter” and “refrest roms” not available  when miyoogamelist.xml is loaded

ac3nl, 2022-09-04:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0 rc

### Miyoo Firmware Version

Latest

### Provide a clear and concise description of the issue

Preciously when the search app was separated, filter and refresh roms options were available at the bottom of the gamelist when miyoogamelist.xml was loaded.

now when this app is integrated, it is not available any more. My arcade list is huge…. Please bring it back

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

Installed 4.0rc.
Refreshed roms

on all gamelists the options are available except for cps1 2 and 3,  and arcade

those have a miyoogamelist.xml file

Aemiii91, 2022-09-04:
The filter function is not supported anymore. Maybe it will be integrated with Onion in a future release.

closed 2022-09-04 by Aemiii91

ac3nl, 2022-09-04:
> The filter function is not supported anymore. Maybe it will be integrated with Onion in a future release.

Should i remove the old app?

Aemiii91, 2022-09-04:
Already answered in https://github.com/OnionUI/Onion/issues/123#issuecomment-1236356001