OnionUI/Onion #123: After cleaning a search, the search app completly disapeared from the menu

ac3nl, 2022-09-04:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0rc

### Miyoo Firmware Version

Latest

### Provide a clear and concise description of the issue

I did a search in the games screen. Cleared the search. Now the search app has disapeared fron the games screen. 

I uninstalled and reinstalled the search app from the package manager but no luck

i can access it from the apps menu though.

how do i bring it back to the games section

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

### Provide any additional context or information that may help us investigate your issue

Searched from games menu
Cleared search

app disapeared completely for games menu

Aemiii91, 2022-09-04:
This is the new behavior of the search app. Not an issue. For easy access you can map search to either X or Y button.

closed 2022-09-04 by Aemiii91

ac3nl, 2022-09-04:
> This is the new behavior of the search app. Not an issue. For easy access you can map search to either X or Y button. 

Should i remove the old app?

Aemiii91, 2022-09-04:
The recommendation is gonna be to "Uninstall filter" in the old search tools, deleting the `App/SearchFilter` folder, then reinstall with Package Manager.

ac3nl, 2022-09-04:
> The recommendation is gonna be to "Uninstall filter" in the old search tools, deleting the `App/SearchFilter` folder, then reinstall with Package Manager.

I updated with searfilter installed. Now i can't find the old app
In de the app menu

Aemiii91, 2022-09-04:
You can just reinstall it (it's still available on GitHub)

ac3nl, 2022-09-04:
> You can just reinstall it (it's still available on GitHub)

Okay that worked.
Maybe in the final version an auto uninstall
If the old app is detected. Now it interferes with the 
New version. If the old app is there, you cannot bind the button correctly the Screen goes black and no search appears

After your advice, the button bind workd correctly.
Although i miss the filter function

Kind regards

Ac3

Aemiii91, 2022-09-04:
Oh yeah, that's right. The old version isn't using the correct display method, and can't be used with the new shortcuts.

Aemiii91, 2022-09-04:
We will look into integrating a filter function in v4.1+

ac3nl, 2022-09-05:
> We will look into integrating a filter function in v4.1+

And the refresh rom option, when you just added a few roms to a single system. This was awesome too, kinda miss that now

Aemiii91, 2022-09-10:
The "Filter" and "Refresh roms" are now included in V4, install Filter via Package Manager.

ac3nl, 2022-09-10:
Really great!! 

Thank you so much!!