OnionUI/Onion #126: Search app launched within 'Games' menu will only show results and won't allow to set search string.

ZoranLain, 2022-09-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.0

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

When launching 'Search' app from the 'Games' menu where all the systems are listed, the Search app will only show the results of the previous search and there is no way to set a new search string. 
This is not an issue when I launch Search app from the Apps menu or from within the Recents menu (yup, I enabled recents).  I will get the normal screen to input search string every time I launch Search in these 2 ways. 

### Steps to Reproduce

I went thru an update from 3.12.3 to 4.0.0.  After that, I noticed the Search App was missing from the Games menu (which is also an issue but not the one I am reporting right now).   I then found Search under Apps menu and launched it from there.  Did a search and it returned a list of results organized in folders by the system as expected.  I went back to the Games menu and the Search app is now there.  The problem is when I launch it, I only get the results list from my previous search instead of getting the search string input screen.  This makes the Search app not usable when launched within the Games menu.

### Provide any additional context or information that may help us investigate your issue

The only workaround I found is to launch Search app from with the Apps menu or the Recents menu, it behaves correctly when launched from those menus.

Aemiii91, 2022-09-10:
This was already discussed here: https://github.com/OnionUI/Onion/issues/123

Please search for your issue before filing a new one!

Aemiii91, 2022-09-10:
Furthermore, the search term can be changed by selecting the bottom list item in the *search result list* found under `Games` -> `Search` (as usual).

closed 2022-09-10 by Aemiii91