OnionUI/Onion #127: Add 'Diagnostics' option to Tweaks

schmurtzm, 2022-09-10:
Hey guys, future feature idea: 'Diagnostics' option in tweaks...
If possible, it would:
- Turn on RA logging* 
- Do a directory print of the SD card
- Make a copy of  essential files (RA/Onion versions, RA config, logs, lists)
- Dump everything into a root diags folder or zip that the user can easily share?

*With logging it may have to be a sort of 2-step process since we want to enable it but only capture the output after they perform the steps to reproduce their issue - needs some thought

schmurtzm, 2022-09-10:
Good idea : I would add an option for dev to desactivate all the logs dump on the SD card : it will allow to have all the output directly on the UART without modifying the runtime.sh file.

XK9274, 2023-06-17:
Will work on this

XK9274, 2023-06-17:
Ok this is what I have below, the vision to use it currently will be to be launched through a `Tweaks -> System -> Diagnostics -> Start logging` (terminal output is just to show the scope). When logging is finished it will be something like: `Tweaks -> System -> Diagnostics -> Stop and export logging` which will stop and archive the logs. 

Currently this doesn't redirect anything from runtime.sh to any log files but it exports some useful information. 

## The script accepts the args:

`logger.sh all` which starts everything

or any of the below:

```
case $arg in
                system_info) log "Executing system_info"; system_info ;;
                dir_listings) log "Executing dir_listings"; dir_listings ;;
                retroarch_start) log "Executing retroarch_start"; retroarch_start ;;
                retroarch_stop) log "Executing retroarch_stop"; retroarch_stop ;;
                export_logs) log "Executing export_logs"; export_logs ;;
                *) log "Invalid logging operation: $arg"; ;;
 esac
```

## The script should be ended with:

`logger.sh export_logs` which will reset the RA config to what it was before enabling logging and then export everything to a .7z

## Starting logging
```
/mnt/SDCARD # .tmp_update/script/logger.sh all
[2023-06-17 14:51:03] Executing all logging operations
[2023-06-17 14:51:03] Writing Onion version
[2023-06-17 14:51:03] Writing CPU details
[2023-06-17 14:51:03] Writing memory info
[2023-06-17 14:51:03] Writing disk info
[2023-06-17 14:51:03] Writing process list
[2023-06-17 14:51:03] Writing dmesg to file
[2023-06-17 14:51:03] Writing wifi state
[2023-06-17 14:51:03] Checking internet access
[2023-06-17 14:51:04] Getting dir list of retroarch
[2023-06-17 14:51:04] Getting dir list of .tmp_update
[2023-06-17 14:51:07] Getting dir list of miyoo dir
[2023-06-17 14:51:07] Getting dir list of App dir
[2023-06-17 14:51:15] Getting dir list of RApp dir
[2023-06-17 14:51:15] Exporting RA config
[2023-06-17 14:51:15] Backing up RA config
[2023-06-17 14:51:15] Enabling logging in RA
```

## Stop and zip
```
/mnt/SDCARD # .tmp_update/script/logger.sh export_logs
[2023-06-17 14:51:26] Executing export_logs
[2023-06-17 14:51:26] Exporting and archiving logs
[2023-06-17 14:51:26] Copying existing .tmp_update/logs to working dir

7-Zip (z) 22.01 (armt) : Copyright (c) 1999-2022 Igor Pavlov : 2022-07-15
 32-bit arm_v:7 thumb:2 locale=C UTF8=- Threads:2

Open archive: /mnt/SDCARD/log_export.7z
--
Path = /mnt/SDCARD/log_export.7z
Type = 7z
Physical Size = 93924
Headers Size = 423
Method = LZMA2:1536k
Solid = +
Blocks = 1

Scanning the drive:
5 folders, 11 files, 2581101 bytes (2521 KiB)

Updating archive: /mnt/SDCARD/log_export.7z

Add new data to archive: 5 folders, 11 files, 2581101 bytes (2521 KiB)

Files read from disk: 11
Archive size: 95512 bytes (94 KiB)
Everything is Ok

[2023-06-17 14:51:30] Restoring original logging values in RA

/mnt/SDCARD #
```

## It will create the a folder called log_dump with the below contents:
```

/mnt/SDCARD # ls -alhR /mnt/SDCARD/.tmp_update/logdump/
/mnt/SDCARD/.tmp_update/logdump/:
total 224
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 .
drwxrwxrwx   10 root     root       32.0K Jun 17 14:56 ..
drwxrwxrwx    2 root     root       32.0K Jun 17 14:56 configs
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 directories
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 network
drwxrwxrwx    2 root     root       32.0K Jun 17 14:56 runtime_logs
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 sysinfo

/mnt/SDCARD/.tmp_update/logdump/configs:
total 192
drwxrwxrwx    2 root     root       32.0K Jun 17 14:56 .
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 ..
-rwxrwxrwx    1 root     root      105.2K Jun 17 14:56 retroarch.cfg

/mnt/SDCARD/.tmp_update/logdump/directories:
total 3680
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 .
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 ..
-rwxrwxrwx    1 root     root        1.6M Jun 17 14:56 appdir.log
-rwxrwxrwx    1 root     root       66.9K Jun 17 14:56 miyoodir.log
-rwxrwxrwx    1 root     root       85.8K Jun 17 14:56 rappdir.log
-rwxrwxrwx    1 root     root      284.8K Jun 17 14:56 retroarch.log
-rwxrwxrwx    1 root     root        1.4M Jun 17 14:56 tmp_update_dir.log

/mnt/SDCARD/.tmp_update/logdump/network:
total 96
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 .
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 ..
-rwxrwxrwx    1 root     root        2.9K Jun 17 14:56 net_summary.log

/mnt/SDCARD/.tmp_update/logdump/runtime_logs:
total 256
drwxrwxrwx    2 root     root       32.0K Jun 17 14:56 .
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 ..
-rwxrwxrwx    1 root     root           8 Jun 17 14:51 asasas
-rwxrwxrwx    1 root     root           0 Jun 17 14:56 batmon.log
-rwxrwxrwx    1 root     root        1.7K Jun 17 14:56 ftp.log
-rwxrwxrwx    1 root     root        6.0K Jun 17 14:56 install.log
-rwxrwxrwx    1 root     root         475 Jun 17 14:56 network.log
-rwxrwxrwx    1 root     root        7.4K Jun 17 14:56 retroarch__2023_06_17__14_56_27.log
-rwxrwxrwx    1 root     root         544 Jun 17 14:56 scummvm.log

/mnt/SDCARD/.tmp_update/logdump/sysinfo:
total 192
drwxrwxrwx    2 root     root       32.0K Jun 17 04:52 .
drwxrwxrwx    7 root     root       32.0K Jun 17 04:53 ..
-rwxrwxrwx    1 root     root       67.7K Jun 17 14:56 dmesg.log
-rwxrwxrwx    1 root     root       16.8K Jun 17 14:56 sys_summary.log
/mnt/SDCARD #

```

Which will then be zipped up and added to a file called:

`log_dump.7z in SD root (/mnt/SDCARD)`

@Olywa 
@schmurtzm 
@Aemiii91 

Can you think of any other useful logs/directory trees/files or other info would be useful to include?

RE: UART 
what do you redirect logs to for you to see it in UART? i don't have one yet

XK9274, 2023-08-02:
In progress on: 

https://github.com/OnionUI/Onion/pull/1090

closed 2023-10-17 by Aemiii91