OnionUI/Onion #128: zip/unzip capabilities in Pac-Man

schmurtzm, 2022-09-10:
To manage the packages available in "Package Manager" in an easier way we will stock the package in zipped format. 
It will allow for example to stock the [swap file from FFplay](https://github.com/bobotrax/ffplay_Miyoo/releases) with a smaller footprint and make easier package shares/import.

schmurtzm, 2022-09-10:
@Aemiii91 do we include additional information about the package in a separate file or the filename will be enough ?

we could imagine a json to include  : 

- an icon (visible in pacman list) 
- a full description / help (with screenshots ?) when you press select

Aemiii91, 2022-09-10:
I think all help/information/previews belong in the wiki. The icons are a bit tricky, and I don't think they're necessary (we could try to use the apps/emus' real icons, placed in `Icons` or `App/[appname]`).

The zip functionality should be implemented just using the filename, and then we need to be able to list the contents of each zip (to check if they're completely installed, etc)

bobotrax, 2022-09-10:
Hi guys,
How to imagine the format of your package?
I think you need a description, even if it's a short one, to explain what it is.
For example ffplay for most people it doesn't tell them anything, but video player we know right away what it is.
We can imagine something like that:
```
ffplay.pak
├── desc
└── SDCARD
    ├── Emu
    │   └── ffplay
    │       ├── bin
    │       │   └── ...
    │       ├── config.json
    │       ├── icon.png
    │       ├── launch.sh
    │       ├── libs
    │       │   └── ...
    │       └── ram.img
    └── Media
        └── Videos
```
So Package Manager reads the content of desc and extracts the content of SDCARD in /mnt/SDCARD/
And if it bothers that the description is in the zip, we can imagine a package in 2 files, ffplay.pak and ffplay.txt with the description in this file.

Aemiii91, 2022-09-10:
The filename is well suited for a short description: `Video Player (ffplay).pak`. Having multiple files defeats the purpose of using a zip file.

bobotrax, 2022-09-10:
Ok, so the files will be in the root of the zip?
```
Video Player (ffplay).pak
├── Emu
│   └── ffplay
│       ├── bin
│       │   └── ...
│       ├── config.json
│       ├── icon.png
│       ├── launch.sh
│       ├── libs
│       │   └── ...
│       └── ram.img
└── Media
	└── Videos
```

Aemiii91, 2022-09-10:
Yeah that's how Pac-Man currently works (except it's not zipped)