OnionUI/Onion #131: Fast Forward not working on GB/GBA on Onion 4.0.0

josecannete, 2022-09-10:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/Sichroteph/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.0

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Hello, thank you so much for your work.

The fast-forward feature (button R) is not working on GB / GBA after fresh install or upgrade to Onion 4.0.0.

### Steps to Reproduce

I did a fresh install of Onion 4.0.0 and noticed that the fast-forward feature (button R) is not working on GB/GBA (I tried it only on Pokemon games, not sure if the same on other games since are the only ROMs I have).

Then, I did a fresh install of Onion 3.12.3 and there it worked normally. After that I tried updating to 4.0.0 following the [upgrade instructions](https://github.com/OnionUI/Onion/wiki/Installation#upgrading) using the Reinstall option and the same problem appeared. Now I'm going back to 3.12.3 again.

### Provide any additional context or information that may help us investigate your issue

.

Olywa, 2022-09-10:
Fast Forward in Onion V4 is activated with Menu+R (not R alone). 
This is as per the shortcut keys defined here: https://github.com/OnionUI/Onion/wiki/Global-Shortcuts

You don't mention the Menu button, only R. Are you saying it is not working for you with Menu+R ?

tomchandotme, 2022-09-11:
The new version of RetroArch in Onion V4 have probably changed GB/GBA's key binding.
I am sure in older version of Onion, you can fast forward by holding R button in GB/GBA.

daisukedd, 2022-09-11:
Exactly, in the old version the user had two fast-forward options and in v4 it has only one, even configuring another button

tomchandotme, 2022-09-11:
Temporary Solution:

1. Open any GB/GBA game
2. Press Menu + Select to open RetroArch Quick Menu
3. Navigate to Control > Port 1 Controls
4. You can see the L2/R2 buttons are unassigned
5. Assign L2/R2 buttons to Fast Forward

P.S. You have to remapping on each core (GB/GBC/GBA)

Aemiii91, 2022-09-11:
This is not a bug, just new defaults. I will add a note to the changelog.

closed 2022-09-11 by Aemiii91