OnionUI/Onion #133: Blank screen when game state set to empty and returning to Main menu

stiivgiaco, 2022-09-14:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

V4.0.1

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

When removing the last game switcher state thus having it set to empty, when trying to go back to Main menu (by Start or B) user is presented with blank screen other than loading back to the Main menu. User has to perform a force power off

### Steps to Reproduce

1. Open game
2. During Gameplay press Game switcher button
3. From Game switcher menu remove state
4. Having no Game switcher state try to return to Main menu using Start or B button

### Provide any additional context or information that may help us investigate your issue

Tested on onion V4.0.0 and V4.0.1

stiivgiaco, 2022-09-14:
Fresh install on a new sd card 256gb formated to fat32
Encountered using PS Rom and GBA Rom

Aemiii91, 2022-09-14:
Could you try removing (moving to another folder) `Saves/CurrentProfile/saves/playActivity.db` and see if the problem persists.

stiivgiaco, 2022-09-14:
not sure if thats a possibility as this will only happen once when you delete the last game state. If i reboot the device it will not stay on the blank screen regardless if there are any states

Aemiii91, 2022-09-14:
So after a reboot, the GameSwitcher exits without problems?

stiivgiaco, 2022-09-14:
yes

Aemiii91, 2022-09-14:
Can you share a video of it happening? Including launching a game and clearing GS. Maybe I can spot if there's anything out of the ordinary.

stiivgiaco, 2022-09-14:
https://mega.nz/file/taAgkAaC#wHhWv-HJdXk3xrwWGRAtbLk_Aq8U1hzLKpl0a4d3IBg

Aemiii91, 2022-09-14:
Thank you! Unfortunately I couldn't spot anything out of the ordinary (except the black screen of course).

lanserver, 2022-09-15:
I did a clean install of 4.0.1 tonight and I'm running into the same problem with SNES and PSX games, but not NES or GBA games. I've tested and encountered the bug with Link to the Past, Super Mario World, Wayne's World (SNES), and Gran Turismo 2.

Replication steps would be to start with an empty game switcher list (not sure if this makes a difference) and launch a SNES game like Link to the Past. After loading into the game, press the menu button once, then press X to remove the game from the game switcher. You should see the "Empty" message in the switcher, then if you press either B or the menu button to return to the games list, it will go to a black screen instead of the games list.

I also tried renaming `playActivity.db`  and restarting. I confirmed that it's properly recreating the file, but I still get a black screen if I repeat the steps above.

Aemiii91, 2022-09-15:
Thank you @lanserver, I'm now able to reproduce! Now we're getting somewhere.

Aemiii91, 2022-09-15:
This is now fixed for next release ([9b6af63](https://github.com/OnionUI/Onion/commit/9b6af6358f542e0f3e90f9fc9984b2da6a137b33)), please install the development build to confirm.

stiivgiaco, 2022-09-15:
in testing

stiivgiaco, 2022-09-15:
I can confirm that the issue is resolved in dev build [9b6af63](https://github.com/OnionUI/Onion/commit/9b6af6358f542e0f3e90f9fc9984b2da6a137b33)

Aemiii91, 2022-09-15:
Thank you @stiivgiaco !

closed 2022-09-15 by Aemiii91

stiivgiaco, 2022-09-15:
Thank you for the quick fix @Aemiii91

Aemiii91, 2022-09-15:
It was a nice challenge

lanserver, 2022-09-15:
Wow, that was fast. I'll give the new build a look later today. Thanks @Aemiii91!