OnionUI/Onion #135: Default Sega MS core has no key mapped for Button 2

anibaldeboni, 2022-09-16:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.1

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

As the tittle says, the default Sega MS core doesn't have a button mapped to MS Button 2

### Steps to Reproduce

Run some game in MS, Black Belt for example, the character doesn't kick, since it requires Button 2 mapped for the function

### Provide any additional context or information that may help us investigate your issue

It's just a matter of RetroArch core configuration, I fixed it in my device but it would be better if it comes configured by default

Olywa, 2022-09-17:
Thanks for reporting. Button 2 appears to be mapped to R2 by default but I agree, should be a face button. 

Not a big MD player but to me the logical mapping  for the Mini should be:

MD buttons 1,2,3 = b, a, x (since 1 and 2 are most commonly used, you'd want those on b & a)
For 6 button pad config (where MD also has x,y,z) these should be y, L2, R2.

Aemiii91, 2022-09-20:
We had erroneously added a `PicoDrive.rmp` file. This will be removed in next release.

closed 2022-09-20 by Aemiii91