OnionUI/Onion #137: PSX game files appear multiple times in list

harun911, 2022-09-20:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.2

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

First of all, thanks for all your great work!
I've updated successfully from Onion OS 3.12.3 (I believe?) to 4.0.2.
I noticed that my PSX games list became a bit wonky, compared to before. I had several multi-disc games in an m3u file, those games no longer appear in the list. Also, I had several games consisting of a single cue file but with multiple bin files. For instance mortal kombat trilogy, which had a single cue file with a reference in it for like around 30 track bin files. Somehow, all the cue files aren't shown anymore and I'm presented with all the track files in my list. 

### Steps to Reproduce

Add a cue file with multiple bin files in the PS folder and navigate to the PSX game list. 

### Provide any additional context or information that may help us investigate your issue

I think you might need to allow cue files to show instead of bin files? Like the other way around? And perhaps allow m3u files to appear again?

Olywa, 2022-09-21:
Hi there, we made a joint design decision between the Onion team with v4 to only show .bin files and hide the .cue files. This was done because you can launch a game with either, but when both are permitted we see many users not happy to have each game listed twice (once for the bin and once for the cue). [V4 supported extensions are here](https://github.com/OnionUI/Onion/wiki/Emulators#sony---playstation-).  

You are on the opposite side of this but also the first to raise it (if we see many more then we may reconsider the options).

**Fortunately, there is are some simple solutions:**

1) We recommend converting your PSX games to CHD (for single disc) or PBP (for multi-disc games). There is a video with the PlayStation section linked above. It really is a simple (and reasonably quick) process and the cleanest way to store your PSX games.

**-Or-**

2) You can change which extensions are permitted to be displayed. This is less recommended since it may be reset if you come to update/reinstall Onion later. To do this, using a text editor like notepad, open the file `\Emu\PSX\config.json` at the bottom of this file you will see the `"extlist":` values, which details all supported extensions. You can insert `cue | CUE` there, save and then insert back in the Mini. You may need to refresh your roms (press `SELECT` from the Games Tab.)

closed 2022-09-21 by Olywa

Aemiii91, 2022-09-22:
@Olywa maybe it would make more sense to have `cue` in the `extlist`, to have consistency across all systems accepting bin/cue - and consistency with the known standard format of bin/cue (almost always you'll have a `cue` if you have a `bin`, even though it's not commonly needed)

Olywa, 2022-09-22:
Hi @harun911 
After some team discussion, we are asking the users. Have your say by [voting here](https://www.reddit.com/r/MiyooMini/comments/xkziuo/how_do_you_store_your_psx_games_onion_user_onion/
).

zebramonkeylizard, 2022-10-10:
There is no vote on that page. That is a poll asking what people use. If you have bin and cue file then only the cue file should be shown as it is common to have many bin file compared to one cue for a game.

Aemiii91, 2022-10-10:
I refer to this for more info on our final stance: https://github.com/OnionUI/Onion/discussions/568#discussioncomment-3812364

dariusdan, 2022-10-12:
> There is no vote on that page. That is a poll asking what people use. If you have bin and cue files then only the cue file should be shown as it is common to have many bin files compared to one cue for a game.

This exactly.
I was using onion os 3.11.2. Castlevania SotN has 1 cue file and 2 bin files (Track1 / Track2). On the PSX system folder, it correctly hides the bin files and only showed the cue file.
Now I upgraded to the 4.0.3 and the PSX system folder shows only the 2 bin files and no cue file when only the .cue should be. A multi track game should not prioritize the bin files.

I will edit the config file [as suggested](https://github.com/OnionUI/Onion/issues/137#issuecomment-1254327531)