OnionUI/Onion #143: Quick menu all are black

hungtcs, 2022-09-26:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Black background when open quick menu.

### Steps to Reproduce

Press select button in main ui or list ui.

Default theme or any other theme.

### Provide any additional context or information that may help us investigate your issue

[image]
[image]

hungtcs, 2022-09-26:
More Step：
1. Turn off and eject memory card
2. Turn on and `Settings -> Factory Reset`
3. Turn off and insert memory card 
4. Turn on and press `Select` button to open quick menu (now the quick menu background is translucent)
5. Switch theme use `Themes` app. (now the quick menu background is black)

----

[image]
[image]

Aemiii91, 2022-09-27:
This issue is apparently theme specific. Since themes are user-generated content, it's not an issue we can address at this moment.

closed 2022-09-27 by Aemiii91