OnionUI/Onion #144: Bad screen synchronization

vanfanel, 2022-09-27:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

2204190828

### Provide a clear and concise description of the issue

I have just received a Miyoo Mini and installed latest Onion on it.

First, let me say I am a graphics system/backend coder: in the FOSS front, I did the DispmanX backend for SDL1.x on the Raspberry Pi, I did the DispmanX backend for RetroArch, did the RetroArch sunxi graphics backend (Allwinner), and a lot of the KMS/DRM backend for SDL2. So I know what I am talking about here. 

The problem here is that the screen refresh is simply wrong: VSYNC is not really working at all. And I mean at all. Have Vsync ON, Threaded Video OFF, and load a NES rom with a well know 1-pixel smooth scrolling sequence like The Legend of Zelda's item list, and observe the screen "jittering" with small periodic hiccups: the sequence is not smooth by any means, no matter what RetroArch video settings are used.

Now, before returning the thing, is there hope for proper VSYNC? Or is it broken at a backend level? You seem to be using that SDL_Dingux thing I had never built. Is there hope for it to have real, working VSYNC at all?

Thanks

### Steps to Reproduce

Load any game with smooth scrolling sequence and play with the video settings.
Whoever reads this on the Onion team knows the issue for sure.

### Provide any additional context or information that may help us investigate your issue

Possibly related to that SDL_Dingux thing and how it manages buffer swapping. But I need confirmation to keep or return the console.

vanfanel, 2022-09-27:
@Aemiii91 I was talking about the LOZ item description screen that appears after the tittle screen if you just boot the game and let it run.

But you can see the same defect in every smooth-scrolling game: for example, try the scrolling test in the 240p suite available here: https://artemiourbina.itch.io/240p-test-suite
The Genesis/MegaDrive version will do, but every system exhibits the same results because the problem is on the video backend.

vanfanel, 2022-09-27:
Alisia Dragoon intro sequence will also exhibit major scrolling problems periodically.
If you can't see that, well, I guess the thing is DOA for me.

Aemiii91, 2022-09-27:
This is a recording of the scroll test in 240p suite (SNES version included with Onion). I don't see any problems, can you provide video sample for comparison? Have you tried using stock or MiniUI?

https://user-images.githubusercontent.com/44569252/192606132-f5f18cfc-ee6e-4de3-857c-233633a4430b.mp4

vanfanel, 2022-09-27:
Recording an MP4 video to illustrate a video refresh problem / lack of problem makes absolutely no sense at all: your recording is not synchronized to the console refresh rate, so there are very visible desyncs in your videos, but that doesn't mean anything.

You can simply look at the RetroArch text scroll animation: does it look right to you?

Anyway, well, I already noticed the Aliexpress seller and asked for a return.

closed 2022-09-27 by Aemiii91