OnionUI/Onion #197: Flashback (REminiscence Core) won't launch by command line

schmurtzm, 2022-09-27:
Hi, I try to create a "launch.sh" to run Flashback Port ([libretro documentation](https://docs.libretro.com/library/reminiscence/)) directly from the new PORTS menu
Result for the moment Retroarch is launched with a blackscreen. If I go in retroarch menu and start flashback manually it works.
I don't know if I 've made an error in my launch.sh : 

Seems OK to me... a bug in the core ? 🤔 

```bash
#!/bin/sh
echo $0 $*
progdir=`dirname "$0"`
homedir=`dirname "$1"`

if [ -f "/mnt/SDCARD/Emu/PORTS/Binaries/Flashback.port/FILE_HERE/level1.lev" ]; then
	# Timer initialisation
	cd /mnt/SDCARD/App/PlayActivity
	./playActivity "init"

	cd /mnt/SDCARD/RetroArch/
	HOME=/mnt/SDCARD/RetroArch/ /mnt/SDCARD/RetroArch/retroarch -v -L /mnt/SDCARD/RetroArch/.retroarch/cores/reminiscence_libretro.so "/mnt/SDCARD/Emu/PORTS/Binaries/Flashback.port/FILE_HERE/level1.lev"

	# Timer registration
	cd /mnt/SDCARD/App/PlayActivity
	./playActivity "Flashback"
else
	cd "/mnt/SDCARD/Emu/PORTS/Binaries/missingFile"
	./infoPanel
fi
```

Here my files (without games content) : 
https://dl.dropbox.com/s/v0cdwqdoirp30kc/flashback.zip

Olywa, 2022-09-27:
@schmurtzm - did you ever figure out for sure if this was a core issue (on PC also?) ?

schmurtzm, 2022-10-04:
It's working fine on PC.
I've updated REminiscence Core with last sources and it still have the same behavior...
But when I take a look on this core issue I see many blackscreen problems similar to ours.
Anyway this core is not very active because there are alternatives to run flashback on other platforms.

Few time ago I have made a little workaround in the launch.sh (it's not up to date for Onion 4 but the most important are the Retroarch command line)  : 

```
#!/bin/sh
echo $0 $*
progdir=`dirname "$0"`
homedir=`dirname "$1"`

if [ -f "/mnt/SDCARD/Emu/PORTS/Binaries/Flashback.port/FILE_HERE/level1.lev" ]; then
	# Timer initialisation
	cd /mnt/SDCARD/App/PlayActivity
	./playActivity "init"

	cd /mnt/SDCARD/RetroArch/
	HOME=/mnt/SDCARD/RetroArch/ /mnt/SDCARD/RetroArch/retroarch -v -L /mnt/SDCARD/RetroArch/.retroarch/cores/reminiscence_libretro.so "/mnt/SDCARD/Emu/PORTS/Binaries/Flashback.port/FILE_HERE/level1.lev" &
	sleep 3s
	pkill -9 retroarch
	HOME=/mnt/SDCARD/RetroArch/ /mnt/SDCARD/RetroArch/retroarch -v -L /mnt/SDCARD/RetroArch/.retroarch/cores/reminiscence_libretro.so --menu

	# Timer registration
	cd /mnt/SDCARD/App/PlayActivity
	./playActivity "Flashback"
else
	cd "/mnt/SDCARD/Emu/PORTS/Binaries/missingFile"
	./infoPanel
fi
```

So it runs RA in background , then close , then open retroarch again
So you just have to go in history and run the last game to start it : not really clean but quick to run :p

schmurtzm, 2022-11-08:
Some additional information : 
https://github.com/libretro/REminiscence/issues/13
We now have a workaround ;)

closed 2023-03-23 by Aemiii91