OnionUI/Onion #280: Favorites list randomly wiped out

polaris911, 2022-09-27:


filipmnowak, 2023-05-28:
> @polaris911 are you on the latest OnionOS version, If you are still running an older version I recommend updating. That said, I've still not seen anyone else have this problem :(

still happens in `Onion V4.2.0.2-beta`. already 2 times within 2 days.

> Are you using either the SD Card or the USB card reader that came with the device as both of these are known to cause problems and can cause corrupt and/or missing files.

i doubt it's SD card issue - i am using high endurance card.
and i doubt it's SD card reader also...

bfaliszek, 2023-07-04:
Still have issue with disappearing games from favorites list. On `Onion V4.2.0-beta-dev`, lates built on Myioo Mini Plus.
I manually edited the .json file with the list of favorite games and for a while the list was ok. After restarting the device, the games disappeared again. The `{}` is missing from the JSON file

Aemiii91, 2023-07-04:
Miyoo has tried to fix this, but it seems it's still happening. We still have no idea how to reproduce.