OnionUI/Onion #560: Problem with default gba emulator

mcfly90, 2022-09-30:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

April 2022

### Provide a clear and concise description of the issue

The gba emulator default gpsp have problems with game like 007 nightfire, colin mcrae 2.0, and others, slow fps and for example with colin mcrae before star a race the game exit and back to menu onion.

The best option is mgba, no problems with this core, please put on default list of games

### Steps to Reproduce

Select colin mcrae or 007 nightfire sith gpsp default emulator core on games.
Try to star a race on colin mcrae
Graphics glitch and sound on 007 nightfire

### Provide any additional context or information that may help us investigate your issue

The core mgba on experimental tab is the best option

Aemiii91, 2022-09-30:
We would love to do this! But I've been told there's some demanding games that run best on gpSP (I have no idea which ones). We would need to do a comprehensive test, testing the most popular and demanding games, and which effect this will have for users' saves.

Aemiii91, 2022-10-01:
I did a quick user inquiry on Discord on which games were most demanding. I have tested each game on both cores with the following result:

[image]

We see no reason not to make the switch.

mcfly90, 2022-10-01:
Thanks, the gpsp is a very bad core

mcfly90, 2022-10-01:
And, if make the change a mgba, to the save files, rename the sav file to srm (mgba), so progress will not be lost on games

Aemiii91, 2022-10-01:
> And, if make the change a mgba, to the save files, rename the sav file to srm (mgba), so progress will not be lost on games

Yes, we need to figure out if we can reliably do this automatically during install.

rollercoaster, 2022-11-05:
Is this something an advanced user can try on their mini with v4.0.3? If yes, how please?

Aemiii91, 2022-11-05:
Edit `Emu/GBA/launch.sh` and replace "`gpsp_libretro`" with "`mgba_libretro`"

rollercoaster, 2022-11-07:
I transferred Mother 3 1.3 save file and it worked after a bit of finagling. 

Two remarked for anyone else trying this:

1. That worked but the save file is created in the saves folder and not the saves/mGBA (as it does for save/gpSP). How do I mGBA to save inside the saves/mGBA folder?
2. On **gpSP**, the file was called "Mother 3 (Japan) [T-En by Chewy & Jeffman & Tomato v1.3] [HQ audio mix by MusicTheorist].sav" where as for **mGBA**, it was called "21. Mother 3 (Japan) [T+Eng].srm" (presumably because mGBA is using the 7z file name and gpSP is using the ROM file name inside the 7z.

schmurtzm, 2022-11-08:
> 2\. presumably because mGBA is using the 7z file name and gpSP is using the ROM file name inside the 7z.

It's not a good new for the automatic import of saves 😞

rollercoaster, 2022-11-08:
Maybe there are config options for the mGBA that we can adjust for the right save folder and the right ROM name? I am a noob in this category so can you point me to the config file or review this yourself please.

andymcca, 2023-01-06:
Please note that in terms of gpSP - we are currently investigating the game issues quoted over at the libretro gpsp repo.  So far Colin McRae Rally 2.0 has been fixed.  Grand Theft Auto also appears to be working (no crash after the menu)?  I will continue working through these games to test and report findings on the repo so that they can be fixed.

closed 2023-03-23 by Aemiii91