OnionUI/Onion #561: Slow performance on Sega 32X after upgrading to V4

aiguanachein, 2022-09-30:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.03

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

Sega 32X games are running very slow after upgrading to V4, they worked fine before.
I tried deleting all the savestates as stated in the wiki, but didn't see any change on performance.

### Steps to Reproduce

Just execute any 32X game

### Provide any additional context or information that may help us investigate your issue

Tried some games all with the same result. Virtua Fighter, Virtua Racing, Tempo, NBA JAM...

Aemiii91, 2022-09-30:
Do you have anything in `Saves/CurrentProfile/config/PicoDrive`?

aiguanachein, 2022-09-30:
Yes, I had these 4 files GG.cfg, Picodrive.cfg, Picodrive.opt and THIRTYTWOX.cfg
I deleted them and now works like a charm! Thanks!!

Aemiii91, 2022-09-30:
Great! I think will make this trick more available somewhere in the wiki 😊

closed 2022-09-30 by Aemiii91