OnionUI/Onion #566: mame2003plus version is too old

ZOMBIE39, 2022-10-05:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

20220419

### Provide a clear and concise description of the issue

The version of mame2003plus that comes with the Onion system is too old and does not support the newly added rom.
If possible, it would be appreciated if a new version of the core could be compiled for the Onion system.

### Steps to Reproduce

https://raw.githubusercontent.com/libretro/mame2003-plus-libretro/master/CHANGELOG.md
Run the new game mentioned in the changelog above.

### Provide any additional context or information that may help us investigate your issue

After checking, the Onion system comes with mame2003plus compiled with the code version of March this year.
A complete rom set based on the latest romset from mame2003plus.
https://pan.baidu.com/s/1BDBUZwOR-vlfNCJ_Na7vTQ?pwd=gggg
Network Cloud Drive Extraction Code：GGGG
I will try to find another network cloud drive to upload the rom if this one is not available for download.
The one below is a rom I tried that an older version of the core was determined not to support.
[xeno.zip](https://github.com/OnionUI/Onion/files/9714521/xeno.zip)

Olywa, 2022-10-06:
Hey there. First thing, please do not post links to rom packs or attach commercially available games here (we do not want any legal issues). I have removed these from your initial post.
-- 
I've dug into this and I believe your issue is more than just with the Mame core version in Onion since Xeno Crisis was supported from the end of last year so our current Mame2003plus build from March should cater for it. In any case, I have tried this game with the Onion V4 default core build and with a recent [1.10.3 build](https://github.com/OnionUI/Onion/files/9719855/mame2003_libretro.zip) compiled just a few weeks ago sadly with no luck. 

I was able to find the following release instructions in relation to running this game in Mame:
 _Create a folder or zip file called xenocrisis with the MAME ROM files and then copy the text from the provided [neogeo.xml](https://github.com/OnionUI/Onion/files/9720096/neogeo.zip) into the neogeo.xml file found in the hash folder inside MAME's root._

However, an [issue logged on the Mame2003Plus github](https://github.com/libretro/mame2003-plus-libretro/issues/1254) seems to dispute the necessity of doing so and the advice there is to place a neogeo bios in the same folder as the Mame rom. In any case I have tried with instructions there and above (trying with various neogeo bios files, including the xml and not), naming the rom itself as both xeno.zip and zenocrisis.zip) but in all cases I have had no success getting this game to run. 

As this is not an Onion OS issue but more with the core, the related configuration, and a unique homebrew game, I cannot spend more time on this. The links/files are there should you wish to explore further yourself (or reach out to someone on the cores own github issue). My gut feeling is it needs a very specific neogeo Bios file to run but I did try several.  

I did have success getting both the MegaDrive and NeoGeoCD versions of this game to run using the default Onion emulators so I would suggest using one of those. The NeoGeoCD is obviously the best version and also has a far better control scheme than the MD port. The NGCD rom comes in a Cue+ISO+Wav format and should be converted into CHD to be recognised by Onions NeoGeoCD core [with CHDMAN using the cue file as the source](https://www.youtube.com/watch?v=DGKTCox6SyY) resulting in a playable 417mb CHD file. 

If you do however figure out the issue with Mame, please feel free to comment back here as it may help others in the future.

closed 2022-10-06 by Olywa

ZOMBIE39, 2022-10-06:
> Hey there. Please do not post links to rom packs or attach commercially available games here (we do not want any legal issues). I have removed these from your initial post.
> 
> -- I've dug into this and I believe your issue is more than just with the Mame core version in Onion since Xeno Crisis was supported from the end of last year so our current Mame2003plus build from March should cater for it. In any case, I have tried this game with the Onion V4 default core build and with a recent [1.10.3 build](https://github.com/OnionUI/Onion/files/9719855/mame2003_libretro.zip) compiled just a few weeks ago sadly with no luck.
> 
> I was able to find the following release instructions in relation to running this game in Mame: _Create a folder or zip file called xenocrisis with the MAME ROM files and then copy the text from the provided [neogeo.xml](https://github.com/OnionUI/Onion/files/9720096/neogeo.zip) into the neogeo.xml file found in the hash folder inside MAME's root._
> 
> However, an [issue logged on the Mame2003Plus github](https://github.com/libretro/mame2003-plus-libretro/issues/1254) seems to dispute the necessity of doing so and the advice there is to place a neogeo bios in the same folder as the Mame rom. In any case I have tried with the instructions there and above (trying with various neogeo bios files, placing the bios with the rom and in the the mame2003_plus bios folder, including the xml and not, naming the rom itself as both xeno.zip and zenocrisis.zip) but in all cases I have had no success getting this game to run.
> 
> As this is not an Onion OS issue but more with the core, the related configuration, and a unique homebrew game, I cannot spend more time on this. The links/files are there should you wish to explore further yourself (or reach out to someone on the cores own github issue). My gut feeling is it needs a very specific neogeo Bios file to run but I did try several.
> 
> I did have success getting both the MegaDrive and NeoGeoCD versions of this game to run using the default Onion emulators so I would suggest using one of those. The NeoGeoCD is obviously the best version and also has a far better control scheme than the MD port. The NGCD rom comes in a Cue+ISO+Wav format and should be converted into CHD to be recognised by Onions NeoGeoCD core [with CHDMAN using the cue file as the source](https://www.youtube.com/watch?v=DGKTCox6SyY) resulting in a playable 417mb CHD file.
> 
> If you do however figure out the issue with Mame, please feel free to comment back here as it may help others in the future.

Sorry, copyright issues for these retro games in China are not very sensitive.
Since I was running the game on the PC version of retroarch with the latest version of mame2003plus core without any additional settings and it worked, I hastily concluded that older versions of mame2003plus may not be supported.
Thank you very much for the useful information regarding this issue and I will try to explore a solution for this game.

ZOMBIE39, 2022-10-11:
@Olywa 
Oh, it turns out that I'd been overlooking a problem, because I'd been using a non-merged set, so I didn't need to put an extra BIOS file to run the MVS game, but ClrMamePro didn't automatically add the BIOS file to the Xeno Crisis zip file after scanning. After putting the BIOS file neogeo.zip into the rom directory, it worked without any additional settings.
I guess it's probably illegal to distribute BIOS files, so I'll provide the CRC of my BIOS file.
[image]
[image]
[image]
[image]

Olywa, 2022-10-11:
Hey there, that's great! Thanks for reporting back, it will be really useful for others who might hit issues with this game. 
Yeah, unfortunately we can't share Bios files here either so really appreciate the screenshots with CRC. 

Just to clarify, you don't have `neogeo.zip` inside `\BIOS\mame2003-plus\` - correct?

- You put the neogeo.zip only in `\Roms\Arcade\ `(as shown)
- You put the rom in `\Roms\Arcade\` (can you confirm how the rom is named and whether you extracted it to a folder)?

Off the bat I can see your `neogeo.zip` has 44 files (mine has only 29) so that could also account for it not running on my device. What is the size of your `neogeo.zip` ?

ZOMBIE39, 2022-10-11:
> Hey there, that's great! Thanks for reporting back, it will be really useful for others who might hit issues with this game. Yeah, unfortunately we can't share Bios files here either so really [...]

Yes, I put it in the rom directory and not the system directory or BIOS directory, this is highlighted in the core instructions of mame2003plus, it has to be in the rom directory.
As for the naming it is xeno.zip, which is a modified file name based on the romset of mame2003plus.
The file size of neogeo.zip is as follows.
[image]
[image]
[image]
[image]

Olywa, 2022-10-11:
Perfect 👌 So for anyone stumbling across this, in short summary, **the solutions is**:

- Place both `xeno.zip` and `neogeo.zip` in ` \Roms\Arcade\`
- The `neogeo.zip` **must** be a compatible Mame 0.78 or 2003-Plus version 

> The one from my Neo Geo rom folder did not work, nor did a couple of others I tested with initially (I did say in my initial response "_My gut feeling is it needs a very specific neogeo Bios file to run_" 😀) .

The `neogeo zip` from "Romset 0.78 Lr Mame 2003" is a also good one, I've just got it running with this (it's only 10 files and 232kb)...

[image]

Thank you again. If you get a chance, do try the NeoGeoCD version ;)

ZOMBIE39, 2022-10-11:
> Perfect 👌 So for anyone stumbling across this, in short summary, **the solutions is**:
>
> * Place both `xeno.zip` and `neogeo.zip` in ` \Roms\Arcade\`
> * The `neogeo.zip` **must** be a compatible Mame [...]

In fact I was thinking at first that maybe you had the wrong version of the BIOS file, not that you needed a special BIOS, but that mame2003plus has its own matching neogeo.zip, which should have more files than mame2003, and that using mame2003plus's romset would tell you the corresponding file needed, what I would suggest is to use this BIOS file to prevent unknown problems.
This file can be obtained by searching for the MAME 2003 Plus Reference Set at https://archive.org/ and is the first item.
As for the extra files I've added, I suppose you can ignore them, but more than that won't cause additional problems, so I always do that.
Here is the data from the original mame2003plus neogeo.zip
[image]
[image]
OK, I'll try the NeoGeoCD version afterwards. 👌

Olywa, 2022-10-11:
Good info. 👍Sorry mate, gonna have to remove the links to that site as it contains roms. 
I did play one an a half levels with that Bios without issue but it's a fair point. Most of our users would not know where to start with ClrMamePro though.

ZOMBIE39, 2022-10-11:
> Good info. 👍Sorry mate, gonna have to remove the links to that site as it contains roms. I did play one and a half levels with that Bios without issue but it's a fair point. Most of our users would not know where to start with ClrMamePro though.

I scanned it with the split sets and found no changes to neogeo.zip, so I've removed the description for this part of ClrMamePro.
ClrMamePro is a handy tool to have when you have the romset file to get the quickest information about the file of your missing roms so you can find the matching roms for the core version.
The BIOS file only works when loading the game, but the core doesn't tell you what files are wrong and what files are missing when it loads like the standalone emulator does, which is why I stress that it's best to use the correct and complete BIOS.

Olywa, 2022-10-11:
I'm very familiar with ClrMamePro (I've been using Mame for 20+ years 😉) but for a lot of Mini users, this is their first Emulation device and the majority just want to be shown what files to use and copy them (or just will just use what came with the device). They are less concerned with curating a perfect romset or using tools 👍

ZOMBIE39, 2022-10-12:
> I'm very familiar with ClrMamePro (I've been using Mame for 20+ years 😉) but for a lot of Mini users, this is their first Emulation device and the majority just want to be shown what files to use [...]

Playing retro games under copyright restrictions is a real pain, as the content organiser cannot put in the BIOS files and newcomers have to fend for themselves to get the files they need. It's also necessary for the content organizer to alert and answer people who have problems with the BIOS and ROMs.
I'm a bit glad I developed an interest in retro gaming in China, where these things are largely unregulated and can be easily exchanged.
It's not manufacturer friendly, but these old games are more widely known because of emulators, and then when manufacturers re-sell these old games or put out spin-offs they get more people to buy them. Emulators also help manufacturers save on development costs, and while this is again not friendly to emulator developers, it's good to be able to go further with each other, I guess🤔.