OnionUI/Onion #572: "[InitKeyShm]: create shm fail" Games not loading kicked back to game selection

BinChickenJosh, 2022-10-09:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

"[InitKeyShm]: create shm fail" Games not loading kicked back to game selection

I have found that adding the default ROMs collection and other ROMs collections (around 20GB) leads to an issue resulting in games not being playable and kicking me back to the menu selection for the console i selected.

I have not found any solution to this issue and have started from scratch with new cards Samsung Evo plus and Sandisk ultra, new cores, new ROMs, and even older copies of onion os such as v3 and vs 4.0.1. however, still remains that MiniUI and Miyoo UI are the only successful frontends.

However I have managed to find a forum with other people experiencing the same issue, and my finding found the string in logs "[InitKeyShm]: create shm fail" and also having less than 99 games seems to fix the issue.

also seems to be only with a select few of games larger games, the smaller ones seem to work**

brought my Miyoo mini v2(v3) last week and have the "latest firmware too" and i have this issue without themes 

Reddit forum -->   https://www.reddit.com/r/MiyooMini/comments/xqbj68/games_not_launching_on_onion_403_and_latest/

### Steps to Reproduce

Please provide detailed steps that will help us replicate and investigate your issue

logs provide information
[install.log](https://github.com/OnionUI/Onion/files/9740740/install.log)
 

and video link (google) for evidence
https://photos.app.goo.gl/JWo42Vkjt2g3AhNc6

### Provide any additional context or information that may help us investigate your issue

very happy with onion os so far love the customization i'm considering donating for the great working efforts

Aemiii91, 2022-10-09:
I will start off by saying, we see this problem often with the stock rom collection, and we recommend to source your roms in another way.

I have thousands of games in each list and no problem. The keyshm fail is just a warning message, it doesn't impact the system.

To try and get to the bottom of this, I would start by asking you to enable RetroArch logging.

Go to `Apps` -> `RetroArch` -> `Settings` -> `Logging` and enable `Log to File`. Then back out to the main RetroArch menu, and go to `Configuration File` and choose `Save Current Configuration`.

Then try to launch some games, and see if anything can be found out by reading `.tmp_update/logs/retroarch.log`.

Aemiii91, 2022-10-09:
The interesting part being:
`[ERROR] [Content]: Could not read content file "/mnt/SDCARD/RApp/snes9x/../../Roms/SFC/Super Mario World (USA).zip"`

Which tells me there's a problem with the rom file, not the system. Have you tried unzipping it?

Aemiii91, 2022-10-09:
Maybe that solved your problem @iconoclusterdotexe, but @BinChickenJosh said on reddit: "Also I have reloaded the games list but still kicks me ?", so I suspect this will not work for him.

Olywa, 2022-10-10:
I would also ask if @BinChickenJosh can confirm that each of the solutions here have been ruled out, it always comes down to one of these things: [I just installed Onion, why do some games no longer launch?](https://github.com/OnionUI/Onion/wiki/Frequently-Asked-Questions-%28FAQ%29#i-just-installed-onion-why-do-some-games-no-longer-launch)

BinChickenJosh, 2022-10-10:
**### Found the solution,** 
it was the ROMs ☹️ I loaded my Mario super circuit Rom in mGBA on my computer and it didn't load either but the original copy from Miyoo mini works fine on stock os and miniUI (seems it bypasses checksum maybe?) but when I play to my pc it shows that it is corrupt instantly. see my log file if you like but it fails checksum and exits, my guess is that the ROMs that I have collected and the ROMs from Miyoo are mostly broken and corrupt, however, I can confirm that I found another copy of Mario super circuit I had in its own .zip file and playing it from the Miyoo mini worked!! I think it is safe to close this thread and publish that Miyoo 's games bypass checksum and are corrupt.? thanks for the suggestions and support and keep up the good effort! 

[corrupt rom.log](https://github.com/OnionUI/Onion/files/9744357/corrupt.rom.log)
[working rom.log](https://github.com/OnionUI/Onion/files/9744359/working.rom.log)

@iconoclusterdotexe 
@Olywa 
@Aemiii91

Aemiii91, 2022-10-10:
Glad you figured it out! It is very strange that the games work on the other operating systems.

closed 2022-10-10 by Aemiii91