OnionUI/Onion #578: Activity Tracker Doesn't Handle Period/Dots in Game Title

cuteseal1, 2022-10-12:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204190828

### Provide a clear and concise description of the issue

When viewing the Activity Tracker, ROMs that have periods / dots / full stops in their file names are not displayed correctly.
The game names are truncated from the last period / dot / full stop onwards.

For example:
01. S.T.U.N. Runner = 01. S.T.U.N
02. Chrono Trigger = 02
03. Ms. Pacman = 03. Ms

### Steps to Reproduce

1. Put ROMs with periods / dots / full stops in their filename in the ROMs folder
2. Open ROMs and play
3. View Activity Tracker

### Provide any additional context or information that may help us investigate your issue

[image]

Olywa, 2022-10-14:
There appears to be a little more to this than just names being truncated after a period as if you look at my example (last game), that's not happening for me. We've also not seen this reported before so I'm wondering if there is something unique going on with your setup.

[image]

cuteseal1, 2022-10-14:
Thank you very much for looking into this. Very interesting.  It so happens I got a new micro sd card today so I did:

1.  A fresh install of onion os
2. Copied my /roms and /saves over
3. In package manager, uninstalled and re-installed the Activity Tracker app
4. Deleted the /saves/playactivity.db file
5. And it looks like it is still happening

All the roms are from the platform explorer pack on reddit. As far as I can tell I can’t see any other special characters in the file name. 

I have also checked that there is no miyoogamelist.xml file in the folder. 

[image]
[image]
[image]

cuteseal1, 2022-10-14:
If you would like to test / this particular file is in the Super Nintendo pack - download link is in row 34:

https://docs.google.com/spreadsheets/u/0/d/1qe3vRUXWJxlsztp21Fpmq-yyWNdo-SzDl4i_PSkzbM0/htmlview

what is unique about this rom pack is that they have compiled the top 100 games for each platform and have named them in order so all the names start with 01. 02. 03. etc. 

Please let me know if there is anything else you would like me to test.

Olywa, 2022-10-14:
Hi there, I have removed the links to roms in your post, that is not permitted here (we do not want to bring legal troubles to the project). 

With that out the way, thanks for all the info, it's useful - we will look a little further into what is happening.

Olywa, 2022-10-15:
@cuteseal1 
OK, I've just done some testing myself and have **confirmed this issue**.
It appears to be system and extension agnostic (I tested with SNES and GBA), using several formats (7zip, zip, sfc, gba). 
It also does not matter whether there is a space or text immediately after the period.
[image]
[image]  [image]

This doesn't explain why Mario vs. Donkey Kong (GBA), per my earlier example, displays correctly 🤷‍♂️

@Aemiii91 I've done all I can, with you for investigation 😉

Olywa, 2022-10-15:
@cuteseal1 This will be investigated but until a solution is identified and released and in order to resolve immediately in your case, I have prepared two Windows scripts to batch edit your rom names which you can use as a workaround...

- The first replaces periods immediately followed by a space, with two spaces (to make it easier to reverse). 
- The second undoes this action (replacing two spaces with a period and space to return everything to how it was. 

[Batch Scritps to remove periods.zip](https://github.com/OnionUI/Onion/files/9790996/Batch.Scritps.to.remove.periods.zip)

**Note**: You will need to edit the path `A:\0-TEMP DOWNLOADS\rom test\` in the batch script (with a text editor like notepad) to replace the path for your actual rom path. If you are not worried about reversing later then you can also edit `set "space=  "` (from two spaces to be just one space).

Maybe do a test with one system first. There will be the odd game that you may have to edit manually (like S.T.U.N. Runner) but it should be good for the majority.  

The issue will remain open until a proper solution is found/implemented.

cuteseal1, 2022-10-15:
Thank you Olywa for confirming - I’m glad I’m not the only one!

And I appreciate the workaround. Hopefully a proper fix is not far off.

Aemiii91, 2022-10-26:
We no longer strip everything after the last dot in the next release.

closed 2022-10-26 by Aemiii91

cuteseal1, 2022-10-26:
Thank you! Looking forward to the next release!