OnionUI/Onion #582: Retroarch fails to save game override file(s)

Wilmatron, 2022-10-17:
### Checklist

- [X] I am reporting a bug in the Onion OS (Note that Onion devs can’t fix inherited issues with RA Cores)
- [X] I have reviewed the [Onion Wiki](https://github.com/OnionUI/Onion/wiki) and did not find the answer
- [X] I have searched the existing (open and closed) issues to ensure I do not log a duplicate

### Onions OS Version

4.0.3

### Miyoo Firmware Version

202204

### Provide a clear and concise description of the issue

Retroarch does not save game specific override file(s) - changes made apply to all games of a particular core.

### Steps to Reproduce

Load game - Enter Retroarch menu - Make changes to the core options - Overrides - Save Game Override file

### Provide any additional context or information that may help us investigate your issue

Retroarch reports the saving of the override file is successful, but I see no evidence of this when browsing the SD card; I see no game specific options file anywhere - and I am familiar with Retroarch and its workings.

This is a pretty big deal as some games (for ZX Spectrum / Commodore) demand specific memory requirements and must be loaded from a game override file to operate properly.

Aemiii91, 2022-10-17:
Just tested this with a SNES game, and it works fine. Game override were saved to `Saves/CurrentProfile/config/Supafaust/Tetris & Dr. Mario (USA).cfg`.

If you still think this is an issue, please be more specific.

Wilmatron, 2022-10-17:
Thanks for testing @Aemiii91 and for your comments. This was a PICNIC error. I had confused the game options file with the game config file. Sorry for wasting your time!

closed 2022-10-17 by Wilmatron